wroberts/rogauracore #4: Make for Mac os

MohammadtaghiFarkhondekar, 2019-04-22:
Hello .
Thank you for nice working .
Can you develope this project for Mac os because we can using the Mac OS on Asus gl553vd and other models with Aura core keyboard .
https://github.com/MohammadtaghiFarkhondekar/macOS-Mojave-For-Asus-ROG-GL553VD
Thank you .

thorhop, 2019-04-24:
Technically it's possible since the project relies on **libusb**. However, this source has been built to work on Linux alone. Can you perhaps try to build the thing? What you'd probably need to do is install the Xcode build environment and also the required dev files. You might need to install **brew** to get libusb-dev on your system.

So.. to recap:
1. Install [XCode](http://developer.apple.com/TOOLS/Xcode/)
2. Install [MacPorts](https://www.macports.org/install.php)
3. Use MacPorts to install `libusb` and `libusb-devel`
3. Wait until bug #5 has been resolved.
4. Try to build **rogauracore** on mac OS.
5. Collect logs from the build to resolve build issues for mac OS

If you can do this we could gauge the possibility of making the
code mac OS compatible. However, I wouldn't hold my breath.
Not sure how much code in this project is Linux-centric. I just
added a push request to add udev support - udev being a Linux
only system. However, I added checks as well, so the udev rules
should be ignored when configuring and building on mac OS. 

We're also pretty much reliant upon @wroberts and if he has the
time to refactor this code to work on mac OS. It might be too big
of a job.

wroberts, 2019-04-24:
Hi @MohammadtaghiFarkhondekar 

Off the top of my head, I can't think of a reason why this project wouldn't work under Mac OS X.  The code is ~550 lines of good old C, and the only dependency is libusb, which would work fine on OS X.  Googling around, I see some recommendations to try `libusb-compat` under homebrew (might also be a similar package for macports).  My last apple notebook died a year or two ago, so I won't be much help in trying to get this to work, but I'll leave this issue open in case anyone else has any news about compiling under OS X.  As @thorhop outlined, you'll need XCode because Apple doesn't ship compilers built into the operating system.  A package manager like homebrew or macports should help with getting libusb-dev, but one could also just build and install this from source, I guess.  Good luck!

hieplpvip, 2019-07-21:
@wroberts Hi Roberts, I've ported your program to macOS. I call it [macrogaura](https://github.com/hieplpvip/macrogaura).

I have listed your name in credits section of README. Would it be ok?