wroberts/rogauracore #5: Current master won't build (autoreconf problem).

thorhop, 2019-04-24:
**OS_release**:
NAME="Pop!_OS"
VERSION="19.04"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Pop!_OS 19.04"
VERSION_ID="19.04"
HOME_URL="https://system76.com/pop"
SUPPORT_URL="http://support.system76.com"
BUG_REPORT_URL="https://github.com/pop-os/pop/issues"
PRIVACY_POLICY_URL="https://system76.com/privacy"
VERSION_CODENAME=disco
UBUNTU_CODENAME=disco

I tested this by cloning the code directly from `wroberts/rogauracore` and trying to build it on 19.04 of Pop!_OS (which technically is not stable just yet, which I'm making clear right now).

I have most build dependencies provided by the Ubuntu meta package.

** Steps to reproduce:**

1. Clone `wlroberts/rogauracore`
2. cd/to/dir
3. Run `autoreconf`
4. Receive errors about missing files in `config/`

**Log**:
```
hobbes :: ~/Sources/rogauracore ‹master› % autoreconf                                                                                                                                    
configure.ac:11: error: required file 'config/compile' not found
configure.ac:11:   'automake --add-missing' can install 'compile'
configure.ac:8: error: required file 'config/install-sh' not found
configure.ac:8:   'automake --add-missing' can install 'install-sh'
configure.ac:8: error: required file 'config/missing' not found
configure.ac:8:   'automake --add-missing' can install 'missing'
Makefile.am: error: required file 'config/depcomp' not found
Makefile.am:   'automake --add-missing' can install 'depcomp'
autoreconf: automake failed with exit status: 1
hobbes :: ~/Sources/rogauracore ‹master› ./configure                                                                                                                                1 ↵
configure: error: cannot find install-sh, install.sh, or shtool in config "."/config
hobbes :: ~/Sources/rogauracore ‹master› make                                                                                                                                       1 ↵
make: *** No targets specified and no makefile found.  Stop.
```

**Issue**:
A proper ./configure and makefile are not generated at all, meaning the source cannot be built.

**Possible solution**:
 Running these commands (requiring **libtool**) would fix the issue, and possibly straighten up some of the automake files...

```
$ libtoolize --force
$ aclocal
$ autoheader
$ automake --force-missing --add-missing
$ autoconf
$ ./configure
$ make
$ sudo make install
```
This will change the make files however, and if that's a good thing or bad thing I'm not sure.

wroberts, 2019-04-24:
Good catch.  Running `autoreconf -i` will fix this.  I've updated the README.

closed 2019-04-24 by wroberts

PedroPaulBermeo, 2019-11-09:
I cant Execute make, please tell me a solve 
[image]