wroberts/rogauracore #8: Not working on Asus GL504GM

nico87, 2019-05-06:
Hi,

I just tried with compiling rogauracore and running it on a GL504GM.
If I run `rogauracore single_static ffffff` it returns `Could not open ROG Aura Core keyboard: LIBUSB_ERROR_ACCESS`.
So I run `sudo rogauracore single_static ffffff`. In this case no error is returned but the backlight color does not change.

May I provide more information on my system that could be helpful for you?

Thank you!

wroberts, 2019-05-06:
Hi @nico87 

Thanks for the ticket!  As you have rightly deduced, you need to have superuser privileges to modify the keyboard RGB state.

Support for the 0x1866 keyboard in your machine was added by PR #2 and works on the GM501, so I'm a little puzzled why it doesn't work on the GL504.

One thing we could try is looking at the verbose output of `rogauracore`.  If possible, try running this:

```
sudo rogauracore -v single_static ffffff
```

Cheers, Will

nico87, 2019-05-06:
Thanks. This is the output. Seems there are no errors, bu unfortunately the keyboard backlight is still in rainbow mode.

```
parse color ffffff
interpreted color 255 255 255
args:
color1 255 255 255
color2 240 54 127
color3 0 0 0
color4 0 0 0
speed 0
single_static
constructed 1 messages:
message 0: 5d b3 00 00 ff ff ff 00 00 00 00 00 00 00 00 00 00 
Initialising libusb
Initialised libusb.
Found 5 USB devices.
Checking device 1d6b:0003, address 1
Checking device 13d3:56a2, address 2
Checking device 8087:0aaa, address 4
Checking device 0b05:1866, address 3
Found ROG Aura Core keyboard.
Opened USB device.
Auto detach kernel mode set.
Claimed interface 0.
```

wroberts, 2019-05-07:
Yeah, that all looks normal.  Maybe the keyboard is somehow different even though it has a product ID we know about,  Can you try:
```
sudo lsusb -v -d 0b05:
```

Another thought: Asus Aura is the name used for a bunch of different control systems.  On Asus motherboards and graphics cards, for instance, it controls RGB lighting in the case (and external LED strips); that hardware isn't connected on USB, so Aura control is via SMBus.  I know Asus laptops also have an internal SMBus, but I've not heard of it being used to control the keyboard lighting.  However, maybe that's what's going on with your machine.  Do you have any RGB lighting that's not on the keyboard?  Multicolour LEDs on the case or the lid perhaps?

nico87, 2019-05-07:
Yes. It has a front led and a led logo on the monitor cover.

```
Bus 001 Device 003: ID 0b05:1866 ASUSTek Computer, Inc. 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x0b05 ASUSTek Computer, Inc.
  idProduct          0x1866 
  bcdDevice            0.02
  iManufacturer           1 ASUSTeK Computer Inc.
  iProduct                2 N-KEY Device
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x005b
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              3 (error)
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      67
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              1 ASUSTeK Computer Inc.
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      65
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              1 ASUSTeK Computer Inc.
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     167
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x04  EP 4 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0001
  Self Powered
```

wroberts, 2019-05-07:
So here's my keyboard:
```
Bus 001 Device 005: ID 0b05:1869 ASUSTek Computer, Inc. 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x0b05 ASUSTek Computer, Inc.
  idProduct          0x1869 
  bcdDevice            0.03
  iManufacturer           1 ITE Tech. Inc.
  iProduct                2 ITE Device(8910)
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           34
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     263
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               4
Device Status:     0x0001
  Self Powered
```

There are some differences to your output: I've only got one interface, whereas you have three.  This could be a good start.   The endpoint under my interface with bInterfaceNumber 0 looks very much like yours, but we already know `rogauracore` is talking to your bInterfaceNumber 0, and it's not doing the trick.  The second interface on your keyboard (with bInterfaceNumber 1) also looks very similar; I wonder if we should try talking to that endpoint?  Your third interface looks different, it has two endpoints, so that seems less promising.  I'm intrigued by the `iInterface 3 (error)` on your first interface, though I guess that's probably not all that important.  Would be curious to know if this value changes, e.g., after a system reboot, but probably it's always 3.

If you think the front LED and lid LED can change colours, that might also support for the idea that your machine uses a different bus to manage RGB.  [Here](https://gitlab.com/CalcProgrammer1/OpenAuraSDK/issues/17) is an issue I filed against OpenAuraSDK while I was thinking that my laptop might support the full Asus Aura spec (like allowing the keyboard flash in time to music and so on).  Sniffing in this direction, you'd want to find the SMBus and modprobe the driver (probably Intel hardware, so it'd be `sudo modprobe i2c-i801`), and then do i2cdetect to see if it looks like there's an RGB controller on the bus.

At this point, I would bet that the most promising avenue would be to try the USB approach again, but attaching to the second interface instead of the first.  I'll try to push a few changes to the Python prototype of rogauracore tonight that you could try out.

nico87, 2019-05-07:
> If you think the front LED and lid LED can change colours, that might also support for the idea that your machine uses a different bus to manage RGB.

Yes! Definitely it's like that. The lid and front LEDs are just like the keyboard.

The following is the `i2cdetect -l` output:

```
i2c-3	i2c       	i915 gmbus dpb                  	I2C adapter
i2c-1	i2c       	Synopsys DesignWare I2C adapter 	I2C adapter
i2c-6	i2c       	i915 gmbus dpd                  	I2C adapter
i2c-4	i2c       	i915 gmbus dpc                  	I2C adapter
i2c-2	i2c       	Synopsys DesignWare I2C adapter 	I2C adapter
i2c-0	smbus     	SMBus I801 adapter at efa0      	SMBus adapter
i2c-7	i2c       	DPDDC-A                         	I2C adapter
i2c-5	i2c       	i915 gmbus misc                 	I2C adapter
```

Thanks for your help!

wroberts, 2019-05-07:
OK, so the `0xefa0` address for the i801 doesn't seem to match up with OpenAuraSDK.  Nonetheless, you could try:
```
sudo i2cdetect 0
```
Maybe the i2c device looks more familiar.

In the mean time, let's try the USB again, but talking to interface 1 instead of interface 0.  There's a script you can download at https://raw.githubusercontent.com/wroberts/rogauracore/nico87/prototype/rogauracore.py .  You'll need Python3 and python3-usb (`sudo apt install python3 python3-usb`), and then you'd do something like `sudo python3 rogauracore.py single_static ff0000`.

nico87, 2019-05-07:
The i2cdetect output is:

```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- 08 -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: 30 -- -- -- -- 35 36 -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- 44 -- -- -- -- -- -- -- -- -- -- -- 
50: 50 -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --   
```
Running the python script returns an exception:
```
Traceback (most recent call last):
  File "rogauracore.py", line 262, in <module>
    main()
  File "rogauracore.py", line 256, in main
    raise e
  File "rogauracore.py", line 253, in main
    dev.ctrl_transfer(bmRequestType, bRequest, wValue, wIndex, message)
  File "/usr/lib/python3/dist-packages/usb/core.py", line 1034, in ctrl_transfer
    self._ctx.managed_claim_interface(self, interface_number)
  File "/usr/lib/python3/dist-packages/usb/core.py", line 102, in wrapper
    return f(self, *args, **kwargs)
  File "/usr/lib/python3/dist-packages/usb/core.py", line 167, in managed_claim_interface
    self.backend.claim_interface(self.handle, i)
  File "/usr/lib/python3/dist-packages/usb/backend/libusb1.py", line 811, in claim_interface
    _check(self.lib.libusb_claim_interface(dev_handle.handle, intf))
  File "/usr/lib/python3/dist-packages/usb/backend/libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 16] Resource busy
```

wroberts, 2019-05-07:
i2c output looks just like mine, so I guess SMBus isn't the answer here.  Sorry about the USB error, I don't quite understand that; I'll need to go and figure out why we can't connect to interface 1.  Let's try interface 0 again, but using the python code.  Updated script at https://github.com/wroberts/rogauracore/blob/nico87/prototype/rogauracore.py

nico87, 2019-05-08:
Thanks for your help.
The new script doesn't show any error but the keyboard backlight nor the other LEDs didn't change. :-/
I'm a dev myself, so if there's something I can try by editing the code you linked, let me know. It could be faster.

Thanks!

wroberts, 2019-05-08:
OK, so the Python version of `rogauracore` doesn't work either; that's not too surprising, since the C is just translated from the Python.  Too bad, I was hoping we would get lucky.

To sum up our story so far: On the GL504GM, and, as of this morning, on the GU501G (#9), `rogauracore` runs without error messages, but the keyboard colours don't change.  These machines both use the `0b05:1866` USB keyboard, which is also included on the GM501.  I have a report from @thorhop that `rogauracore` works on that laptop (#2), and that all colour modes are functional.

At this point, I don't have a solid hunch for why things are working on one laptop model but not on the two others.  [Asus's website](https://www.asus.com/Laptops/ROG-Strix-Hero-II/) says the GL504GM supports Aura Sync, which my GL503 does not support, so there might be a different control system or a different protocol being used.  The GL504GM also has a programmable RGB light bar along the front of the laptop chassis, presumably not controlled by the keyboard.  I can't find any current information on the GU501.

I've muddied the waters a little bit by talking about i2c and SMBus; this thinking was in relation to the [OpenAuraSDK project](https://gitlab.com/CalcProgrammer1/OpenAuraSDK), which aims to support Asus's Aura and Aura Sync products under Linux, but which at the moment mostly seems to be focussed on RGB control of RAM modules on Asus motherboards.  The hardware targeted by OpenAuraSDK is a dedicated Aura control chip on the motherboard, which is addressable over SMBus.  As far as I can see, this SMBus seems to live at address 0x0b00 or 0x0b20 on AMD hardware, and possibly at 0xf000 on Intel.  If you query the bus, you'll see a list of devices on the bus that are present on the machine.  The Aura chip might live at 0x4e.  You can see RAM sticks under 0x5 (0x50 is the first RAM stick, 0x51 the second, etc.).  Where there is colour control of the RAM, there will be corresponding Aura controllers on 0x70, 0x71, etc.  There is also the [OpenPyAura](https://gitlab.com/thelastguardian/openpyaura) project that is working on similar themes, might be interesting to check out.  These projects are quite young, but are more sophisticated in terms of reverse engineering the Aura protocol, documenting different registers and settings, and supporting a much wider range of hardware than `rogauracore`.

To be clear, `rogauracore` only communicates on USB, and does not do anything with i2c or SMBus.  In this respect, it's similar to other projects to control RGB keyboards such as the Razer Chroma and Corsair K70.  [KeyboardVisualizer](https://github.com/CalcProgrammer1/KeyboardVisualizer) is, I think, the project where I stumbled onto some of this material.

All of this is just background to our central question, which we still don't have an answer to: how is it that the 0b05:1866 USB keyboard can be controlled on the GM501 but not on the GU501 or GL504?  Do these two machines use USB for RGB control?  If so, do they use a different interface or a different protocol than the GM501?  If not, do they use an Aura control chip like Asus desktop motherboards do, and is this controlled over the SMBus?

I suggest we engage in some sanity testing here.  Questions:

1. Do any of the colour commands in `rogauracore` work?  We have established that `single_color` doesn't work; how about the others?

2. Do you have a Windows dual-boot?  If so, can you install Lighting Services and Aura Core (or Aura Sync?) on the Windows, and does it control the RGB there?  If yes, can you list the different colour modes and parameters that the software allows you to control?  Are they the same as the set that `rogauracore` supports?

nico87, 2019-05-08:
Thanks for the clarification. I'll give OpenPyAura a shot later.

>> Do any of the colour commands in rogauracore work? We have established that single_color doesn't work; how about the others?

None of them work

> Do you have a Windows dual-boot? If so, can you install Lighting Services and Aura Core (or Aura Sync?) on the Windows, and does it control the RGB there? If yes, can you list the different colour [...]

Unfortunately, I don't have a dual boot. I don't have the possibility to mount another nvme ssd either and try it :-(
I need Linux for my daily work so I cannot erase it and start from scratch.

stuarttempleton, 2019-05-08:
I'm going to take a look at manipulating it from within VirtualBox to see if I can get any info.

stuarttempleton, 2019-05-11:
I wanted to follow-up. I'm not having any luck getting it to work via virtualbox. I'm not sure how to get it to actually see the USB keyboard -- I'm assuming user error for that. I'm not sure how to proceed, but I"ll keep an eye on this thread.

nico87, 2019-05-11:
To be able to connect USB devices, make sure you installed the vbox
additions and that your user is in the vboxusers group. Then, logout and
login again to make sure changes are effective.

VesterDe, 2019-09-18:
Something strange happened to me, and I thought maybe sharing it would help.

This is on the Strix GL703GM.
OS is Arch Linux.

I had the same problem as every other currently open issue on this topic. Then I loaded Vbox and edited my lights from within there.

Sidenote for Arch (maybe other?) users: I had to install virtuabox-ext-oracle (and reboot after) and also on the Windows image in Virtualbox go to Preferences -> USB -> checkbox on USB 2.0, before it worked.

Once I changed the lights a bit from the box, I downloaded Wireshark to start snooping the USB traffic, but just randomly also ran the rogauracore command again on my host machine. Suddenly, the lights changed colour...

I have no idea why, but it works now... so yay?

The output from the command is the exact same as it was before, when it didn't work.
```
sudo ./rogauracore -v single_static ffffff    
                                                                                                                                                                    
parse color ffffff
interpreted color 255 255 255
args:
color1 255 255 255
color2 26 174 127
color3 0 0 0
color4 0 0 0
speed 0
single_static
constructed 1 messages:
message 0: 5d b3 00 00 ff ff ff 00 00 00 00 00 00 00 00 00 00 
Initialising libusb
Initialised libusb.
Found 6 USB devices.
Checking device 1d6b:0003, address 1
Checking device 13d3:56a2, address 3
Checking device 248a:8566, address 2
Checking device 8087:0aaa, address 5
Checking device 0b05:1866, address 4
Found ROG Aura Core keyboard.
Opened USB device.
Auto detach kernel mode set.
Claimed interface 0
```
I hope this helps with the detective work.

BenLand100, 2019-11-19:
I had a similar experience to @VesterDe except I couldn't get the AuraSync software installed in my VM - simply having booted the Windows VM with the device passed through seemed to enable something and allow rogauracore to work.

JoshDreamland, 2019-11-24:
I'm noticing a similar behavior. Linux starts up, my keyboard backlight shuts off. It's as though it has to be switched into "accept lighting commands" mode. When rebooting, the rainbow comes on and stays on, with animation, right until GRUB loads up. Then it's dead. There must be some message that turns the system on. How did you uncover the formats you've already documented, Will?

JoshDreamland, 2019-11-27:
Never mind. It was just the brightness. I added brightness controls to the application and was able to use it exactly as you had it.

closed 2019-12-04 by wroberts

wantedrbk, 2020-04-09:
Hi can i used it for Arch ( Manjaro KDE ) ?

xit4, 2020-04-09:
@wroberts why was this issue closed?
It was not fixed as far as I understood from the messages and I myself have a `GL504GM` with keyboard `0b05:1866` and running any command results in a correct execution but no light changes on the keyboard.
```
sudo rogauracore -v black
args:
color1 48 7 12
color2 160 60 127
color3 0 0 192
color4 10 12 160
single_static
constructed 1 messages:
message 0: 5d b3 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Initialising libusb
Initialised libusb.
Found 6 USB devices.
Checking device 1d6b:0003, address 1
Checking device 13d3:56a2, address 3
Checking device 045e:076c, address 2
Checking device 8087:0aaa, address 5
Checking device 0b05:1866, address 4
Found ROG Aura Core keyboard.
Opened USB device.
Auto detach kernel mode set.
Got configuration descriptor.
Found 3 interfaces on the USB device.
Claimed interface 0.
Successfully sent all messages.
Sent SET message.
Sent APPLY message.
```

kbaleo, 2020-06-05:
Hi,
I had the same issue on UBUNTU 20.04 with a GM501
I did chmod +s /usr/local/bin/rogauracore
Afterwards I did 
rogauracore initialize_keyboard
followed by 
rogauracore single_static ff0000
All works!

thorhop, 2020-06-26:
> Hi,
> I had the same issue on UBUNTU 20.04 with a GM501
> I did chmod +s /usr/local/bin/rogauracore
> Afterwards I did
> rogauracore initialize_keyboard
> followed by
> rogauracore single_static ff0000
> All works!

Can confirm. Works on my Zephyrus M GM501GM as well.

PR #25 (2019-12-04): Add support for brightness adjustment.
I believe this will fix #8 and possibly #23. This fixes the tool on my system, which is actually a G731GW, but behaves similarly to the other open issues I saw.

I had to change some stuff around to accommodate brightness neatly in the CLI. See the commit logs for a more detailed breakdown.

Note that not all color profiles work on my system. Your rainbow effect seems to be a CMYR kind of thing, but my system uses a special message to enable the rainbow effect. Basically, only the static colors work out-of-box for me, even after this PR.

Without this PR, [... cut]