wroberts/rogauracore #9: Difficulty with GU501G

stuarttempleton, 2019-05-08:
I compiled and ran rogauracore. No errors and everything looks reasonable to me, but the colors aren't actually changing. I noticed the PR for the GM501 had my vendor and product id, but that my smbus is showing up on a different i2c. (7 instead of 0). I'm not sure if this is helpful, but here's some more information. Is there anything else I can provide that might help add support for this device?

```shell
sudo i2cdetect -l
i2c-3	i2c       	i915 gmbus dpd                  	I2C adapter
i2c-1	i2c       	i915 gmbus dpc                  	I2C adapter
i2c-6	i2c       	Synopsys DesignWare I2C adapter 	I2C adapter
i2c-4	i2c       	DPDDC-A                         	I2C adapter
i2c-2	i2c       	i915 gmbus misc                 	I2C adapter
i2c-0	i2c       	i915 gmbus dpb                  	I2C adapter
i2c-7	smbus     	SMBus I801 adapter at efa0      	SMBus adapter
i2c-5	i2c       	Synopsys DesignWare I2C adapter 	I2C adapter
```

```shell
sudo i2cdetect 7
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- 08 -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: 30 -- -- -- -- 35 36 -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- 44 -- -- -- -- -- -- -- -- -- -- -- 
50: 50 -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
```
It's interesting that the current code actually finds the device in the libusb device list, but for whatever reason that message doesn't cause a color change. 

```shell
 sudo ./rogauracore -v single_static ff0000
parse color ff0000
interpreted color 255 0 0
args:
color1 255 0 0
color2 232 28 127
color3 0 0 128
color4 3 0 0
speed 0
single_static
constructed 1 messages:
message 0: 5d b3 00 00 ff 00 00 00 00 00 00 00 00 00 00 00 00 
Initialising libusb
Initialised libusb.
Found 5 USB devices.
Checking device 1d6b:0003, address 1
Checking device 13d3:56a2, address 2
Checking device 8087:0aaa, address 4
Checking device 0b05:1866, address 3
Found ROG Aura Core keyboard.
Opened USB device.
Auto detach kernel mode set.
Claimed interface 0.
```

If I update iface to 7 in the py script, just to check, it throws a nasty error.
```shell
sudo python3 ./rogauracore.py single_static ff0000
Traceback (most recent call last):
  File "./rogauracore.py", line 262, in <module>
    main()
  File "./rogauracore.py", line 244, in main
    dev, iface, active_iface = acquire_usb()
  File "./rogauracore.py", line 80, in acquire_usb
    raise e
  File "./rogauracore.py", line 75, in acquire_usb
    usb.util.claim_interface(dev, iface)
  File "/usr/lib/python3/dist-packages/usb/util.py", line 205, in claim_interface
    device._ctx.managed_claim_interface(device, interface)
  File "/usr/lib/python3/dist-packages/usb/core.py", line 102, in wrapper
    return f(self, *args, **kwargs)
  File "/usr/lib/python3/dist-packages/usb/core.py", line 167, in managed_claim_interface
    self.backend.claim_interface(self.handle, i)
  File "/usr/lib/python3/dist-packages/usb/backend/libusb1.py", line 811, in claim_interface
    _check(self.lib.libusb_claim_interface(dev_handle.handle, intf))
  File "/usr/lib/python3/dist-packages/usb/backend/libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 2] Entity not found
```

I feel like this might be overkill on the information, but I was really excited to see this project and I'm happy to help debug on my hardware if you come up with anything to test!

stuarttempleton, 2019-05-08:
Thanks for trying to help add support for my device! I think it is similar to that issue, and I'm not entirely sure how to validate that, though.

1. Yes, I can confirm. None of the options work on my hardware.
2. I do not have a dual boot system. When I did have Windows on this machine, I was able to change the keyboard colors. Right now they're set to blue, for example.
3. I'm not sure about Aura Sync. Aura Core was what I had factory installed.
4. Not that I'm aware of. Note: the backlight on the lid is actually the backlight from the screen and not a color changing LED.

I'll follow bug #8.

joeg1484, 2019-11-03:
I have a gu501 as well and the program doesnt error, but doesnt change. For example... I want to do single static blue:
sudo rogauracore -v single_static 0000ff

parse color 0000ff
interpreted color 0 0 255
args:
color1 0 0 255
color2 0 0 0
color3 0 0 128
color4 3 0 0
speed 896
single_static
constructed 1 messages:
message 0: 5d b3 00 00 00 00 ff 00 00 00 00 00 00 00 00 00 00 
Initialising libusb
Initialised libusb.
Found 7 USB devices.
Checking device 152d:0578, address 2
Checking device 1d6b:0003, address 1
Checking device 13d3:56a2, address 3
Checking device 046d:c53f, address 2
Checking device 8087:0aaa, address 5
Checking device 0b05:1866, address 4
Found ROG Aura Core keyboard.
Opened USB device.
Auto detach kernel mode set.
Claimed interface 0.