wroberts/rogauracore #10: Windows version

0xCTF, 2019-05-09:
Hi,
i was thinking since i bought my GL553VD,2 years ago, to build a desktop software with new effects
like the one when the charger plugged in when the pc in sleep mode, <like a wave>
but i don't know how to start
i learned  the basics of C and JAVA
do you think i can control the rgb leds using JAVA ?
is this project have new effects other than those in aura core ?

Thanks

wroberts, 2019-05-10:
Hi @0xCTF thanks for the message!

`rogauracore` is a small command line program in vanilla C whose only dependency is libusb.  [Libusb is supported on Windows](https://github.com/libusb/libusb/wiki/Windows), so I think you would be able to get this working on Windows.  It doesn't support any colour modes that aren't in aura core, though (it's just a reverse-engineered version of aura core that will work on Linux and OS X).  Furthermore, I don't know of a way right now to change the sleeping effect on the keyboard, or otherwise change the built-in colour effects, so I don't think I can help in those directions.  As for doing this with Java, this is certainly possible; off the top of my head, I can imagine either using Java to call the `rogauracore` binary as a child process, refactoring `rogauracore` to be a command-line program with a linked dynamic library and attaching to that library from Java at runtime, or rewriting the entire program (which is small) in Java using some Java-based USB library.  Hope that helps!

0xCTF, 2019-05-11:
Thank you @wroberts, that's really help

i will see what i can do about that, i will start working on it

Thanks again and good job !

closed 2019-05-11 by 0xCTF