wroberts/rogauracore #11: Any chance to support TUF Gaming FX505DU?

yaliv, 2019-07-08:
I've patched the **i2c-piix4** module by following these guides:
https://gitlab.com/CalcProgrammer1/KeyboardVisualizer/issues/85#note_121577579
https://www.reddit.com/r/linux/comments/asxhcc/i_wrote_a_program_to_control_asus_aura_rgb/egy7uvk/

And here's the result of calling sudo `i2cdetect -l`:
```
i2c-3	smbus     	SMBus PIIX4 adapter port 2 at 0b00	SMBus adapter
i2c-10	i2c       	NVIDIA i2c adapter 5 at 1:00.0  	I2C adapter
i2c-1	i2c       	NVIDIA GPU I2C adapter          	I2C adapter
i2c-8	i2c       	dmdc                            	I2C adapter
i2c-6	smbus     	SMBus PIIX4 adapter port 1 at 0b20	SMBus adapter
i2c-4	smbus     	SMBus PIIX4 adapter port 3 at 0b00	SMBus adapter
i2c-2	smbus     	SMBus PIIX4 adapter port 0 at 0b00	SMBus adapter
i2c-0	i2c       	Synopsys DesignWare I2C adapter 	I2C adapter
i2c-9	i2c       	NVIDIA i2c adapter 1 at 1:00.0  	I2C adapter
i2c-7	i2c       	AMDGPU DM i2c hw bus 0          	I2C adapter
i2c-5	smbus     	SMBus PIIX4 adapter port 4 at 0b00	SMBus adapter
```

I've called `chmod 777` to all the SMBus PIIX4 adapters (`/dev/i2c-6`, etc.).
Then I tried using apps like [Borealis](https://github.com/Philpax/borealis), [OpenAuraSDK](https://gitlab.com/CalcProgrammer1/OpenAuraSDK), but still no luck.
My keyboard backlight is only **red** now.

Here's my HW probe result:
https://linux-hardware.org/?probe=27bb2f0366
There I see my i2c_piix4 vendor is **AMD** and my keyboard is **PS/2**, not USB.

wroberts, 2019-07-16:
Hi @yaliv thanks for the issue report!

I'm not sure how to help you further with your issue.  rogauracore and openauranb both control keyboard RGB over a USB interface only.  If you don't have the keyboard on USB at all, we probably can't help too much.   It's possible that your hardware is controlled over the SMBus, in which case the OpenAuraSDK project might be able to help further.  Good luck!

yaliv, 2019-08-04:
After switching to Linux kernel version 5.3, my keyboard backlight is automatically changing colors now.

closed 2019-08-04 by yaliv