wroberts/rogauracore #12: Add ability to dim backlight

mkreisel, 2019-07-14:
First: this project is awesome! I couldn't find any way to change the default rainbow setting on my G531GT until I found this project.

Could you add support for dimming the backlight?

wroberts, 2019-07-16:
Hi @mkreisel thanks for the issue report!

Backlight is normally controlled by the kernel, and you shouldn't need rogauracore to change the brightness.  This project just aims to support setting the keyboard colours.  Two caveats.  One: for me, rogauracore breaks backlight control until I reboot the machine (or restore from suspend).  Two: you can approximate dimming by scaling down the RGB colour you give rogauracore (like setting single_static to 990000 instead of ff0000).

cshendye, 2019-09-06:
i am getting this error
[image]