wroberts/rogauracore #13: Not working with GM501GS

sudoWiz, 2019-07-24:
Installed via the first method, able to enter commands without errors (sudo rogauracore single_colorcycle 1) but nothing happens on the keyboard. tried all command options, none change the keyboard LEDs at all.

wrapper, 2019-11-15:
+1 same problem on GU502GU

JoshDreamland, 2019-11-28:
Can you try changing the brightness using #25? Patch it and run ``sudo rogauracore brightness 3``.

sudoWiz, 2019-12-05:
Tried this, patched to 1.1 and ran the command, but it just prints the usage list. `brightness` does not appear to be a valid command.

JoshDreamland, 2019-12-05:
The changes are now in master, so you shouldn't need to patch anymore. That said, are you sure you ran your newly-built binary and not the original you had already installed?

sudoWiz, 2019-12-05:
definitely sure,  re-installed my whole OS a few days ago and just installed this from scratch today.

JoshDreamland, 2019-12-05:
And you're sure you built the right branch...? That doesn't make sense... Will was able to test it just fine (although it didn't work on his system because the message isn't supported). You can try patching #26, instead, I guess; it has exactly the code I'm using in it.

sudoWiz, 2019-12-05:
Positive. I built 1.1, the newest available release, following the exact instructions listed in the readme.

JoshDreamland, 2019-12-05:
Oh, you mean literally the latest GitHub release. That is from April. Please build master directly.

sudoWiz, 2019-12-06:
Ah. Got any tips for a git noob? I've never done that before, I've only ever downloaded from the releases tab and worked from the .deb or tar.gz that I find there...

If it helps, I'm on Elementary OS 5.1

JoshDreamland, 2019-12-06:
Oh, I see. Download [the repo as a zip](https://github.com/wroberts/rogauracore/archive/master.zip), unpack it, and open a terminal in it, then just run these:

```bash
autoreconf -i
./configure
make
sudo make install
```

That should put the latest changes on your system.

sudoWiz, 2019-12-06:
Tried that and got this error on make:

~/Documents/rogauracore-master$ make
make  all-am
make[1]: Entering directory '/home/wiz/Documents/rogauracore-master'
depbase=`echo src/rogauracore.o | sed 's|[^/]*$|.deps/&|;s|\.o$||'`;\
gcc -DHAVE_CONFIG_H -I.     -g -O2 -MT src/rogauracore.o -MD -MP -MF $depbase.Tpo -c -o src/rogauracore.o src/rogauracore.c &&\
mv -f $depbase.Tpo $depbase.Po
src/rogauracore.c:278:52: error: initializer element is not constant
     {"single_breathing", &single_breathing, 2, 1, {SPEED}},
                                                    ^~~~~
src/rogauracore.c:278:52: note: (near initialization for ‘FUNCTION_RECORDS[1].scalars[0]’)
src/rogauracore.c:279:54: error: initializer element is not constant
     {"single_colorcycle", &single_colorcycle, 0, 1, {SPEED}},
                                                      ^~~~~
src/rogauracore.c:279:54: note: (near initialization for ‘FUNCTION_RECORDS[2].scalars[0]’)
src/rogauracore.c:281:50: error: initializer element is not constant
     {"multi_breathing", &multi_breathing, 4, 1, {SPEED}},
                                                  ^~~~~
src/rogauracore.c:281:50: note: (near initialization for ‘FUNCTION_RECORDS[4].scalars[0]’)
src/rogauracore.c:291:44: error: initializer element is not constant
     {"brightness", &set_brightness, 0, 1, {BRIGHTNESS}},
                                            ^~~~~~~~~~
src/rogauracore.c:291:44: note: (near initialization for ‘FUNCTION_RECORDS[14].scalars[0]’)
Makefile:395: recipe for target 'src/rogauracore.o' failed
make[1]: *** [src/rogauracore.o] Error 1
make[1]: Leaving directory '/home/wiz/Documents/rogauracore-master'
Makefile:278: recipe for target 'all' failed
make: *** [all] Error 2

And then this error when i tried sudo make install afterward:

~/Documents/rogauracore-master$ sudo make install
[sudo] password for wiz:              
depbase=`echo src/rogauracore.o | sed 's|[^/]*$|.deps/&|;s|\.o$||'`;\
gcc -DHAVE_CONFIG_H -I.     -g -O2 -MT src/rogauracore.o -MD -MP -MF $depbase.Tpo -c -o src/rogauracore.o src/rogauracore.c &&\
mv -f $depbase.Tpo $depbase.Po
src/rogauracore.c:278:52: error: initializer element is not constant
     {"single_breathing", &single_breathing, 2, 1, {SPEED}},
                                                    ^~~~~
src/rogauracore.c:278:52: note: (near initialization for ‘FUNCTION_RECORDS[1].scalars[0]’)
src/rogauracore.c:279:54: error: initializer element is not constant
     {"single_colorcycle", &single_colorcycle, 0, 1, {SPEED}},
                                                      ^~~~~
src/rogauracore.c:279:54: note: (near initialization for ‘FUNCTION_RECORDS[2].scalars[0]’)
src/rogauracore.c:281:50: error: initializer element is not constant
     {"multi_breathing", &multi_breathing, 4, 1, {SPEED}},
                                                  ^~~~~
src/rogauracore.c:281:50: note: (near initialization for ‘FUNCTION_RECORDS[4].scalars[0]’)
src/rogauracore.c:291:44: error: initializer element is not constant
     {"brightness", &set_brightness, 0, 1, {BRIGHTNESS}},
                                            ^~~~~~~~~~
src/rogauracore.c:291:44: note: (near initialization for ‘FUNCTION_RECORDS[14].scalars[0]’)
Makefile:395: recipe for target 'src/rogauracore.o' failed
make: *** [src/rogauracore.o] Error 1

JoshDreamland, 2019-12-06:
Ugh. I wondered about that. I think it wants SPEED and BRIGHTNESS to be enum values or macros instead of const int. I thought that was no longer a problem in modern C installations. I'll prepare a patch.

JoshDreamland, 2019-12-06:
Actually, I am so confused. Of all the things gcc should have taken exception to, that isn't one of them. Would you mind trying the same procedure with [this zip](https://github.com/JoshDreamland/rogauracore/archive/retool.zip)?

sudoWiz, 2019-12-06:
Tried it with that zip.

Results:

~/Downloads/rogauracore-retool$ make
make  all-am
make[1]: Entering directory '/home/wiz/Downloads/rogauracore-retool'
depbase=`echo src/rogauracore.o | sed 's|[^/]*$|.deps/&|;s|\.o$||'`;\
gcc -DHAVE_CONFIG_H -I.     -g -O2 -MT src/rogauracore.o -MD -MP -MF $depbase.Tpo -c -o src/rogauracore.o src/rogauracore.c &&\
mv -f $depbase.Tpo $depbase.Po
src/rogauracore.c:272:61: error: initializer element is not constant
 const ArgumentDef COLOR =  {"COLOR",  AK_COLOR, parseColor, NO_DEFAULT};
                                                             ^~~~~~~~~~
src/rogauracore.c:272:61: note: (near initialization for ‘COLOR.defaultValue’)
src/rogauracore.c:273:61: error: initializer element is not constant
 const ArgumentDef COLOR1 = {"COLOR1", AK_COLOR, parseColor, NO_DEFAULT};
                                                             ^~~~~~~~~~
src/rogauracore.c:273:61: note: (near initialization for ‘COLOR1.defaultValue’)
src/rogauracore.c:274:61: error: initializer element is not constant
 const ArgumentDef COLOR2 = {"COLOR2", AK_COLOR, parseColor, NO_DEFAULT};
                                                             ^~~~~~~~~~
src/rogauracore.c:274:61: note: (near initialization for ‘COLOR2.defaultValue’)
src/rogauracore.c:275:61: error: initializer element is not constant
 const ArgumentDef COLOR3 = {"COLOR3", AK_COLOR, parseColor, NO_DEFAULT};
                                                             ^~~~~~~~~~
src/rogauracore.c:275:61: note: (near initialization for ‘COLOR3.defaultValue’)
src/rogauracore.c:276:61: error: initializer element is not constant
 const ArgumentDef COLOR4 = {"COLOR4", AK_COLOR, parseColor, NO_DEFAULT};
                                                             ^~~~~~~~~~
src/rogauracore.c:276:61: note: (near initialization for ‘COLOR4.defaultValue’)
src/rogauracore.c:277:61: error: initializer element is not constant
 const ArgumentDef SPEED =  {"SPEED", AK_SCALAR, parseSpeed, NO_DEFAULT};
                                                             ^~~~~~~~~~
src/rogauracore.c:277:61: note: (near initialization for ‘SPEED.defaultValue’)
src/rogauracore.c:278:75: error: initializer element is not constant
 const ArgumentDef BRIGHTNESS = {"BRIGHTNESS", AK_SCALAR, parseBrightness, NO_DEFAULT};
                                                                           ^~~~~~~~~~
src/rogauracore.c:278:75: note: (near initialization for ‘BRIGHTNESS.defaultValue’)
src/rogauracore.c:280:70: error: initializer element is not constant
 const ArgumentDef COLOR2_OR_AUTO = {"COLOR2", AK_COLOR,  parseColor, AUTO};
                                                                      ^~~~
src/rogauracore.c:280:70: note: (near initialization for ‘COLOR2_OR_AUTO.defaultValue’)
src/rogauracore.c:281:70: error: initializer element is not constant
 const ArgumentDef SPEED_OR_TWO   =  {"SPEED", AK_SCALAR, parseSpeed, TWO};
                                                                      ^~~
src/rogauracore.c:281:70: note: (near initialization for ‘SPEED_OR_TWO.defaultValue’)
src/rogauracore.c:282:70: error: initializer element is not constant
 const ArgumentDef SPEED_OR_THREE =  {"SPEED", AK_SCALAR, parseSpeed, THREE};
                                                                      ^~~~~
src/rogauracore.c:282:70: note: (near initialization for ‘SPEED_OR_THREE.defaultValue’)
src/rogauracore.c:286:9: error: initializer element is not constant
         COLOR
         ^~~~~
src/rogauracore.c:286:9: note: (near initialization for ‘FUNCTION_RECORDS[0].args[0]’)
src/rogauracore.c:289:9: error: initializer element is not constant
         COLOR1, COLOR2_OR_AUTO, SPEED_OR_TWO
         ^~~~~~
src/rogauracore.c:289:9: note: (near initialization for ‘FUNCTION_RECORDS[1].args[0]’)
src/rogauracore.c:289:17: error: initializer element is not constant
         COLOR1, COLOR2_OR_AUTO, SPEED_OR_TWO
                 ^~~~~~~~~~~~~~
src/rogauracore.c:289:17: note: (near initialization for ‘FUNCTION_RECORDS[1].args[1]’)
src/rogauracore.c:289:33: error: initializer element is not constant
         COLOR1, COLOR2_OR_AUTO, SPEED_OR_TWO
                                 ^~~~~~~~~~~~
src/rogauracore.c:289:33: note: (near initialization for ‘FUNCTION_RECORDS[1].args[2]’)
src/rogauracore.c:292:9: error: initializer element is not constant
         COLOR, SPEED
         ^~~~~
src/rogauracore.c:292:9: note: (near initialization for ‘FUNCTION_RECORDS[2].args[0]’)
[... 162 lines cut ...]
         COLOR1, COLOR2, COLOR3, COLOR4, SPEED
                         ^~~~~~
src/rogauracore.c:301:25: note: (near initialization for ‘FUNCTION_RECORDS[5].args[2]’)
src/rogauracore.c:301:33: error: initializer element is not constant
         COLOR1, COLOR2, COLOR3, COLOR4, SPEED
                                 ^~~~~~
src/rogauracore.c:301:33: note: (near initialization for ‘FUNCTION_RECORDS[5].args[3]’)
src/rogauracore.c:301:41: error: initializer element is not constant
         COLOR1, COLOR2, COLOR3, COLOR4, SPEED
                                         ^~~~~
src/rogauracore.c:301:41: note: (near initialization for ‘FUNCTION_RECORDS[5].args[4]’)
src/rogauracore.c:304:9: error: initializer element is not constant
         SPEED_OR_THREE
         ^~~~~~~~~~~~~~
src/rogauracore.c:304:9: note: (near initialization for ‘FUNCTION_RECORDS[6].args[0]’)
src/rogauracore.c:307:9: error: initializer element is not constant
         BRIGHTNESS
         ^~~~~~~~~~
src/rogauracore.c:307:9: note: (near initialization for ‘FUNCTION_RECORDS[7].args[0]’)
src/rogauracore.c: In function ‘printFuncUsage’:
src/rogauracore.c:355:9: warning: format not a string literal and no format arguments [-Wformat-security]
         printf(arg->name);
         ^~~~~~
Makefile:395: recipe for target 'src/rogauracore.o' failed
make: *** [src/rogauracore.o] Error 1

JoshDreamland, 2019-12-07:
All right; I made it build with C99 strict, and it still didn't give me those errors, so I just preemptively rewrote that area using macros. It's a little less type-safe, now; I'm kind of bringing out the worst in C, that way. But it should work for you, now. [Same zip, different version](https://github.com/JoshDreamland/rogauracore/archive/retool.zip), if you'll humor me for just a little longer. 😉

Thanks for your patience in getting this working.

iammarco11, 2019-12-17:
Great patch @JoshDreamland it works like a charm. It could be helpful if you gave commands to control brightness as well :smile:

JoshDreamland, 2019-12-17:
Not sure what you mean, @iammarco11; adding those brightness commands was the whole point of the original patch. Are they not what you used to get your keyboard working?

iammarco11, 2019-12-17:
Nah nevermind I didn't know the commands...maybe you can make a better README giving examples of each command, it would help many people :smile: