wroberts/rogauracore #14: Not working with GL704GW

trbaldwin60, 2019-08-10:
This is not working with my CUK ASUS ROG Strix Scar II GL704GW Gaming Laptop running Ubuntu Disco Dingo 19.04.
It is so close to your GL703 listed as supported, so how can this be tweaked to get it to work?

trbaldwin60, 2019-08-10:
Here's my lsusb output:
Bus 001 Device 005: ID 0b05:1866 ASUSTek Computer, Inc. 
Perhaps I am not doing something right?

trbaldwin60, 2019-08-10:
Here's an example with verbose turned on, and the keyboard just cycles colors with no change:
sudo ./rogauracore single_static ff0000
parse color ff0000
interpreted color 255 0 0
args:
color1 255 0 0
color2 81 33 127
color3 0 0 0
color4 0 0 0
speed 0
single_static
constructed 1 messages:
message 0: 5d b3 00 00 ff 00 00 00 00 00 00 00 00 00 00 00 00 
Initialising libusb
Initialised libusb.
Found 9 USB devices.
Checking device 1d6b:0003, address 1
Checking device 13d3:56a2, address 3
Checking device 046d:c52b, address 4
Checking device 0557:8021, address 2
Checking device 8087:0aaa, address 6
Checking device 0b05:1866, address 5
Found ROG Aura Core keyboard.
Opened USB device.
Auto detach kernel mode set.
Claimed interface 0.

wroberts, 2019-08-10:
Hi @trbaldwin60 thanks for the bug report!

This device 0b05:1866 is giving a lot of trouble.  Your bug report sounds almost identical to #8 and #9 which are on machines that have the same model of keyboard as yours.  As far as I know, there hasn't been any progress on those two issues since they were reported.  Part of the difficulty is that we'll need somebody with that hardware who can try out solutions and run tests (I can't help with this part, since my keyboard is a different model).

My hunch about your keyboard is that it is controlled over USB (not over i2c or smbus), but that the protocol used to control the RGB is somehow different from the protocol used on other keyboards.  The way to both confirm this hypothesis and also to get `rogauracore` working with your hardware would be to sniff the USB protocol while using a working implementation (i.e., the windows Aura Core software).  If this sounds like something you might be interested in having a look at, you'll find some details on the other two issue reports.  Basically, you would put a windows system into a virtual machine (VirtualBox) and then get the asus software to install on that.  You can then connect the laptop's keyboard to the virtual machine over USB and watch the aura core protocol using wireshark.  Not entirely trivial, but it's how I got `rogauracore` to work on my laptop.