wroberts/rogauracore #15: Not Working in G531-GT

parasjain-12, 2019-08-21:
I have installed via the second method. Here is the output of the following things which I have done it . 
`lsusb`

Output is :
```

Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 0b05:1866 ASUSTek Computer, Inc. 
Bus 001 Device 003: ID 8087:0aaa Intel Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

```
When I run  `sudo ./rogauracore -v  single_static ffffff`

This i what I get:

Output:

```
parse color ffffff
interpreted color 255 255 255
args:
color1 255 255 255
color2 171 153 127
color3 0 0 128
color4 3 0 0
speed 0
single_static
constructed 1 messages:
message 0: 5d b3 00 00 ff ff ff 00 00 00 00 00 00 00 00 00 00 
Initialising libusb
Initialised libusb.
Found 4 USB devices.
Checking device 1d6b:0003, address 1
Checking device 0b05:1866, address 2
Found ROG Aura Core keyboard.
Opened USB device.
Auto detach kernel mode set.
Claimed interface 0.
```

When I ran `sudo lsusb -v -d 0b05:`

Output:

```
Bus 001 Device 002: ID 0b05:1866 ASUSTek Computer, Inc. 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x0b05 ASUSTek Computer, Inc.
  idProduct          0x1866 
  bcdDevice            0.02
  iManufacturer           1 ASUSTeK Computer Inc.
  iProduct                2 N-KEY Device
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           91
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              3 (error)
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      67
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              1 ASUSTeK Computer Inc.
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      65
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              1 ASUSTeK Computer Inc.
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     167
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x04  EP 4 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
Device Status:     0x0001
  Self Powered

```

I have also tried many different solutions present on the internet like` echo 0 | sudo tee /sys/class/leds/asus::kbd_backlight/brightness` But  None worrked .

Please Please help me . Any help is appreciated

wroberts, 2019-08-23:
Duplicate of #14 and #8 and #9

qgnox, 2019-09-03:
@wroberts, I dont know if this can help, I got the GA502 laptop that has the same usb keyboard ids  0x0b05,0x1866 but it only has one color (white) so the only thing that it can do on Windows is increase or decrease brightness (turnoff), and trying to make it work on linux I was able to record a usb cap with a windows pe + the asus services [usbpcap2.zip](https://github.com/wroberts/rogauracore/files/3571897/usbpcap2.zip), just hitting the fn + up and down arrows. 

I tried to replicate what it does in the python version and c version of your repo but in both I get errors of resource busy or libusb io error, but I dont know much how usb works, maybe you can see something to what "endpoint" "hid id" is hitting, I suppose it can also help for the colors issue on the others laptops.

onyxnz, 2019-10-04:
I (oh joy!!) have this working on a new G531GT with EndeavorOS (Archlinux). Maybe it is a permissions issue at the OS level, as I had to sudo rogauracore single_breathing 00ff00 ffffff 1 for example.

onyxnz, 2019-10-04:
Verbose:
parse color 00ff00
interpreted color 0 255 0
parse color ffffff
interpreted color 255 255 255
parse speed 1
args:
color1 0 255 0
color2 255 255 255
color3 0 0 0
color4 0 0 0
speed 1
single_breathing
constructed 1 messages:
message 0: 5d b3 00 01 00 ff 00 e1 00 01 ff ff ff 00 00 00 00 
Initialising libusb
Initialised libusb.
Found 5 USB devices.
Checking device 1d6b:0003, address 1
Checking device 0b05:1866, address 3
Found ROG Aura Core keyboard.
Opened USB device.
Auto detach kernel mode set.
Claimed interface 0.

lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 0b05:1866 ASUSTek Computer, Inc. N-KEY Device
Bus 001 Device 004: ID 8087:0aaa Intel Corp. 
Bus 001 Device 002: ID 056e:00f1 Elecom Co., Ltd ELECOM TrackBall Mouse
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

uname -a
Linux onyx-pc 5.3.1-arch1-1-ARCH #1 SMP PREEMPT Sat Sep 21 11:33:49 UTC 2019 x86_64 GNU/Linux

onyxnz, 2019-11-14:
Of note on my machine, sudo rogauracore rainbow does not produce a rainbow, rather the keys are lit red. AND the under-frame lighting is turned off, which for me is good.

hutchgrant, 2019-12-10:
Can confirm master branch works without issue. Nice job

ni3mm4nd, 2019-12-31:
Hi

I've G531-GU with Ubuntu 19.10 and kernel 5.3.0-24-generic and it's not working :-(

```
$ sudo ./rogauracore -v single_static ffffff
parse color ffffff
interpreted color 255 255 255
args:
color1 255 255 255
color2 254 195 127
color3 0 0 192
color4 26 155 254
single_static
constructed 1 messages:
message 0: 5d b3 00 00 ff ff ff 00 00 00 00 00 00 00 00 00 00 
Initialising libusb
Initialised libusb.
Found 4 USB devices.
Checking device 1d6b:0003, address 1
Checking device 0b05:1866, address 2
Found ROG Aura Core keyboard.
Opened USB device.
Auto detach kernel mode set.
Got configuration descriptor.
Found 3 interfaces on the USB device.
Claimed interface 0.
Successfully sent all messages.
Sent SET message.
Sent APPLY message.
```

```
$ lsusb 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 0b05:1866 ASUSTek Computer, Inc. N-KEY Device
Bus 001 Device 003: ID 8087:0aaa Intel Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

Even if command seems to be successfull nothing happens.

Does anybody have some advice?

Thank you

onyxnz, 2019-12-31:
> Hi
>
> I've G531-GU with Ubuntu 19.10 and kernel 5.3.0-24-generic and it's not working :-(
>
> ```
> $ sudo ./rogauracore -v single_static ffffff
> parse color ffffff
> interpreted color 255 255 255
> args: [...]
Works for me :
sudo ./rogauracore -v single_static ffffff
[sudo] password for onyx: 
parse color ffffff
interpreted color 255 255 255
args:
color1 255 255 255
color2 0 0 0
color3 0 0 128
color4 3 0 0
speed 896
single_static
constructed 1 messages:
message 0: 5d b3 00 00 ff ff ff 00 00 00 00 00 00 00 00 00 00 
Initialising libusb
Initialised libusb.
Found 5 USB devices.
Checking device 1d6b:0003, address 1
Checking device 0b05:1866, address 3
Found ROG Aura Core keyboard.
Opened USB device.
Auto detach kernel mode set.
Claimed interface 0.
[onyx@onyx-pc rogauracore]$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 0b05:1866 ASUSTek Computer, Inc. N-KEY Device
Bus 001 Device 004: ID 8087:0aaa Intel Corp. 
Bus 001 Device 002: ID 056e:00f1 Elecom Co., Ltd ELECOM TrackBall Mouse
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

How about updating to the latest kernel? As I am running Archlinux my kernel is bleeding edge, Ubuntu 19.10 is on 5.3.xx

uname -a
Linux onyx-pc 5.4.6-arch3-1 #1 SMP PREEMPT Tue, 24 Dec 2019 04:36:53 +0000 x86_64 GNU/Linux

ni3mm4nd, 2020-01-01:
> Works for me :
> sudo ./rogauracore -v single_static ffffff
> [sudo] password for onyx:
> parse color ffffff
> interpreted color 255 255 255
> args:
> color1 255 255 255
> color2 0 0 0
> color3 0 0 128
> color4 3 0 0 [...]

Does not work even after kernel upgrade:

```
$ sudo ./rogauracore -v single_static ffffff
[sudo] password for kadlec: 
Reading argument 2 (single_static)
parse color ffffff
Interpreted color 255 255 255
args:
rgb(255, 255, 255)[unspecified][unspecified][unspecified][unspecified][unspecified][unspecified][unspecified]single_static
constructed 1 messages:
message 0: 5d b3 00 00 ff ff ff 00 00 00 00 00 00 00 00 00 00 
Initialising libusb
Initialised libusb.
Found 4 USB devices.
Checking device 1d6b:0003, address 1
Checking device 0b05:1866, address 2
Found ROG Aura Core keyboard.
Opened USB device.
Auto detach kernel mode set.
Got configuration descriptor.
Found 3 interfaces on the USB device.
Claimed interface 0.
Successfully sent all messages.
Sent SET message.
Sent APPLY message.
$ uname -sr
Linux 5.5.0-050500rc4-generic
$ 
```

parasjain-12, 2020-01-01:
> Hi
>
> I've G531-GU with Ubuntu 19.10 and kernel 5.3.0-24-generic and it's not working :-(
>
> ```
> $ sudo ./rogauracore -v single_static ffffff
> parse color ffffff
> interpreted color 255 255 255
> args: [...]

Same here , Everything runs but nothing is happening