wroberts/rogauracore #16: Opening calculator on execution of script

sanjujr, 2019-08-30:
[image]
When trying to execute of a script, gcalc opens up

wroberts, 2019-09-09:
That is strange.  I'm guessing that the keyboard color doesn't change?

enteryournamehere, 2019-09-10:
I have the exact same issue! Using Linux Mint 19.2 Cinnamon on an Asus ROG Strix GL503VD.
This happens on all commands except `rogauracore black`. The command does still work, it just also opens the calculator, after changing the keyboard color.

chshbh, 2019-10-27:
Same issue on Manjaro. Asus ROG Strix GL503VD.

enteryournamehere, 2019-12-05:
A second or so after running the command and changing the keyboard colour there comes an input event for the calculator button from my keyboard. I haven't found any consistent difference between me actually pressing the (fn+)calculator button and the input event form this program with xinput test-xi2. If it's useful for anyone trying to debug this I can post the logs of course.
For anyone still having this issue, a workaround to make it stop opening the calculator is just removing the Calculator button as a shortcut to open the calculator in your keyboard settings - I've changed it to Shift+Calculator and it works just fine, without rogauracore opening it.

wroberts, 2020-12-02:
@CYPHER0041 also reports this issue, also on a GL503GE.