wroberts/rogauracore #23: Not working on GU502GU

wrapper, 2019-11-15:
Was able to install but the commands don't do anything to the lighting on the keyboard

cjkoral, 2019-12-12:
Bumping for this issue since it is my hardware as well. @wroberts  Can you give a brief description of how you went about the initial discovery to get this working for your machine? I would be happy to add this work myself however I have limited experience in this specific area so explicit direction would be a great help. I am unsure how to begin going about accumulating the information necessary from my machine/asus armory crate/aura to properly tackle this.  Thanks for making this to begin with, 1 step closer.

JoshDreamland, 2019-12-19:
@cjkoral Can you confirm that the changes now in master (the new brightness commands) do not enable you to activate your keyboard?

cjkoral, 2019-12-19:
@JoshDreamland I just tried these changes and they unfortunately do not work either

JoshDreamland, 2019-12-19:
Do you have a working Windows installation? The best way to get this working is to boot Windows, launch Wireshark and start recording, then install/launch ArmouryCrate. End the recording after your keyboard is initialized, and that should be enough for me to figure out what's going on. If you're good, you can isolate the messages being passed when you, eg, change colors or switch to a different built-in mode. For example, I did a recording pass where I just cycled between all the modes, and then cycled between rainbow and static color five or six times. I also did a recording where I upped the brightness to max, then lowered it to min, for about four cycles.

Also, just to be sure, have you tried just setting the color to red and the brightness to 3? That's the only thing that seems to reliably work across all models so far.

cjkoral, 2019-12-20:
@JoshDreamland Yes i do have a working windows installation, thats how i'm currently adjusting my keyboard settings. I will get back to you with a wireshark dump of ArmouryCrate.  Setting color to red and brightness to 3 is not functioning on my machine either. Thank you very much for getting back to me here, itd be great to get all this working

cjkoral, 2020-05-20:
Update here, my GU502GU now works after the initialize keyboard functionality was added.  The rainbow functionality does not work at this point, but i believe someone else opened up an issue for that at this time.  Thank you for all your work

cjkoral, 2020-05-21:
Another update, the rainbow function from https://github.com/wroberts/rogauracore/pull/26 works for me.  Any timeline on getting that merged in? I have to hop back and forth between tools at the moment

JoshDreamland, 2020-05-21:
We'll need to backport the existing rainbow messages to it, as well as the new init command. It turns out these models just vary widely in both default implementation and support, so we'll need to offer both until we can accurately detect which one your keyboard supports. We can probably do that detection by serial number if we start compiling that list.

I might find time to do some of that this weekend, but I can't make any promises.

Abn0n, 2020-06-23:
Im new to linux, just started using Zorin, I have the exact same laptop, but I have no clue how to go ahead further to installing this program, can someone help? I copy pasted the first line that it says for the ubuntu install as i'm guessing it should work since it is debian, but the lines after that don't, can someone help? Would really apprieciate simple step by step help for this, as stupid and baby-like as possible. I'm very new to linux. Thank you in advance for having the patience to even respond to this.