wroberts/rogauracore #24: Running it causes power management to be unable to adjust brightness

willlovesbearz, 2019-11-19:
After running rogauracore, the address of the device in systemd changes. upower does not know how to handle this and continues trying to use the now incorrect address.
The workaround I have found for this is running ```sudo systemctl restart upower.service``` afterwards.

enteryournamehere, 2019-11-20:
I have the same issue on Linux Mint 19.2 / linux kernel 5.3.7 on my GL503VD. Thanks for the workaround, I didn't know about that yet!

willlovesbearz, 2019-11-20:
> I have the same issue on Linux Mint 19.2 / linux kernel 5.3.7 on my GL503VD. Thanks for the workaround, I didn't know about that yet!

You're welcome, it took around 5 hours of messing around for me to find that one.

wroberts, 2019-12-04:
This is awesome, @willlovesbearz 
Thanks so much for the tip.  I think I'll include it into the README file for more visibility.

wroberts, 2019-12-04:
OK, it's on the front page now with commit 242616625b63b8c0ca3d25b0c8c9ed9a19c93315.  Thanks again!

closed 2019-12-04 by wroberts