wroberts/rogauracore #27: Not working on GX502GW

niconavarrete, 2019-12-01:
Anything we can make to make it work? lspsci lshw or something?

Thanks,
Nico

flukejones, 2019-12-21:
I have a similar laptop - GX502 (same?). From what I can tell, the command bits are the same. So somewhere it seems to be breaking. I'll try and send some commands while wireshark is open on Linux.

flukejones, 2019-12-21:
Some of the captures here. I didn't actually capture one function (like static) then stop though sorry.
[capture.zip](https://github.com/wroberts/rogauracore/files/3991174/capture.zip)

flukejones, 2019-12-22:
I split the captures here. It looks like the host sends regular packets when on non-static setting? Or is this something else sending them?

[gx502-rgb-wireshark.zip](https://github.com/wroberts/rogauracore/files/3992052/gx502-rgb-wireshark.zip)

niconavarrete, 2019-12-30:
Well, thanks to others, after init up in windows (throught virtualbox) it works as expected ...

(so don't uninstall windows to put linux until you have setup the keyboard)

closed 2019-12-30 by niconavarrete