wroberts/rogauracore #28: Error installing on G531GT-BQ002T i

Deziigner, 2019-12-07:
I was installing via the second method in **elementary OS 5.1 Hera**. on running the "make" command I had this particular issue:

>make  all-am
make[1]: Entering directory '/home/aditya/Documents/rogauracore'
depbase=`echo src/rogauracore.o | sed 's|[^/]*$|.deps/&|;s|\.o$||'`;\
gcc -DHAVE_CONFIG_H -I.     -g -O2 -MT src/rogauracore.o -MD -MP -MF $depbase.Tpo -c -o src/rogauracore.o src/rogauracore.c &&\
mv -f $depbase.Tpo $depbase.Po
src/rogauracore.c:278:52: error: initializer element is not constant
     {"single_breathing", &single_breathing, 2, 1, {SPEED}},
                                                    ^~~~~
src/rogauracore.c:278:52: note: (near initialization for ‘FUNCTION_RECORDS[1].scalars[0]’)
src/rogauracore.c:279:54: error: initializer element is not constant
     {"single_colorcycle", &single_colorcycle, 0, 1, {SPEED}},
                                                      ^~~~~
src/rogauracore.c:279:54: note: (near initialization for ‘FUNCTION_RECORDS[2].scalars[0]’)
src/rogauracore.c:281:50: error: initializer element is not constant
     {"multi_breathing", &multi_breathing, 4, 1, {SPEED}},
                                                  ^~~~~
src/rogauracore.c:281:50: note: (near initialization for ‘FUNCTION_RECORDS[4].scalars[0]’)
src/rogauracore.c:291:44: error: initializer element is not constant
     {"brightness", &set_brightness, 0, 1, {BRIGHTNESS}},
                                            ^~~~~~~~~~
src/rogauracore.c:291:44: note: (near initialization for ‘FUNCTION_RECORDS[14].scalars[0]’)
Makefile:395: recipe for target 'src/rogauracore.o' failed
make[1]: *** [src/rogauracore.o] Error 1
make[1]: Leaving directory '/home/aditya/Documents/rogauracore'
Makefile:278: recipe for target 'all' failed
make: *** [all] Error 2

JoshDreamland, 2019-12-07:
Sorry, I apparently just broke that, but I can't get my GCC to reproduce the error no matter what build flags I specify. Can you verify that #26 builds for you?

Deziigner, 2019-12-08:
I checked #26 . I am a complete noob bro. I didn't get it, what to do. There are 5-6 commits, which one to incorporate?

JoshDreamland, 2019-12-08:
Oh, sorry for the confusion. The changes are available [in this ZIP archive](https://github.com/JoshDreamland/rogauracore/archive/retool.zip). If you wouldn't mind just unpacking that and running the same build commands you were using before, I'd appreciate it. ☺

The tool in that Zip should work, but Will and I are having trouble figuring out how to associate firmware versions with USB messages, so there's a good chance the rainbow and multi_breathing commands will not work for you. I'll send Will a patch for the changes in master if you're able to so much as build it.

Thanks!

Deziigner, 2019-12-09:
Hey, thanks for making it easy. I downloaded the archive and proceded as before and this time there was no error in `make` command and `sudo make install` displayed:
make[1]: Entering directory '/home/aditya/rogauracore-retool'
 /bin/mkdir -p '/usr/local/bin'
  /usr/bin/install -c rogauracore '/usr/local/bin'
make[1]: Nothing to be done for 'install-data-am'.
make[1]: Leaving directory '/home/aditya/rogauracore-retool'

Then I tried `rogauracore blue` and got this:

Could not open ROG Aura Core keyboard: LIBUSB_ERROR_ACCESS

JoshDreamland, 2019-12-09:
Excellent. Thanks for the update! Unfortunately, since it interfaces with USB devices directly, rogauracore needs to run as root (ie, you have to call `sudo rogauracore blue` instead of just `rogauracore blue`).

Also, if nothing happens, try `sudo rogauracore brightness 3`.

We should probably update that error message to suggest running as root...

Deziigner, 2019-12-09:
Hey, thanks bro. It's working now. `rogauracore --help` showed the command structures. Thanks a lot again.

closed 2019-12-09 by Deziigner