wroberts/rogauracore #29: Issues on G531GW (ROG Strix Scar III)

enigma6174, 2019-12-10:
I recently installed Ubuntu 19.10 on my Asus Strix Scar III (Model Number G531GW) and the lighting sometimes works and sometimes it doesn't. My keyboard id is **0b05:1866**
Following is a complete list of issues I have tracked so far (I don't know which ones will be useful):

- Just after installing Ubuntu 19.10, my charger was not being detected anymore. At the same time I installed **rogauracore** which didn't work. 

- I was able to fix the charger problem by long pressing the power button (from Windows 10) for 35 seconds (note that for this to work, the charger needs to be disconnected from the laptop first; **keep a note of this, it is going to be useful for the lighting later on**) but the lighting still did not work; at that point of time I had only tried few of the options.

- I did not use my laptop for 24 hours (was not connected to power source all this while; was at 100% charge though when I last switched off) and this time when I switched on my laptop the lighting surprisingly worked (no idea how) but I was only getting RED; tried changing the colors but did not work

- Later on the same day I noticed an unusual behavior : the colors were actually changing after applying for example `sudo rogauaracore green` but for that to work, I had to disconnect the charger first, power off the laptop, wait for 2-3 minutes, again switch it on and THEN connect the charger (any change of sequence of the events would not make it work; a restart also does not work; it needs to be powered off)

- During this entire experimentation phase of 4 days just once it happened that the light worked but the keyboard did not work (a few hard power offs solved it).

- Finally again after 34 hours of rest (no power source to laptop; battery at 100% charge) the commands finally started working! I mean proper working (you type the command and the lighting immediately changes). However any **breathing** and **rainbow** commands were not working under any circumstances. All others worked perfectly like a charm (I mean type the commands and change the colors).

- Yesterday, I connected my Samsung Galaxy Buds over Bluetooth and everything went wrong from there : the lighting stopped, the charger stopped, the WiFi stopped. Disconnecting and switching off Bluetooth and restarting the computer fixed the WiFi but not the other 2 issues.

- I retraced all the above steps many times and got the charger working in the first attempt but not the lighting system. Now finally the lighting system works but only **white** and **red** and that too I need to power off my computer, disconnect the charger, wait for 2-3 minutes and switch it back on. None of the other colors are working.

_My BIOS settings : UEFI boot enabled (tweaking this doesn't affect the lighting system in my experiments, cannot say for sure), FAST BOOT disabled_

_I know this is NOT A HARDWARE ISSUE or any issue with the internals of my laptop because everything works well on Windows no questions asked so either it is to do with un-supported hardware or with the inherent issues of Ubuntu 19.10 iteself_

_However, if it worked once, why is not working now and more importantly, how do I make it work?_

BogdanDarius, 2019-12-11:
Intresting facts!
I own a ROG Strix Hero III, I didnt have these kind of problems.
For me only the rainbow and multi_static commands are not working.
I am using Arch Linux with this kernel ```5.4.2-arch1-1```.
What kernel do you use?

enigma6174, 2019-12-12:
Right now even the red is not working for me anymore. Only the white light is working (well, at least the lights are working!) and my kernel is **5.3.0-24-generic**

BogdanDarius, 2019-12-13:
Have you tried other kernels?

enigma6174, 2019-12-17:
No. I haven't. Which one do you suggest?

BogdanDarius, 2019-12-17:
As I told you I use ``5.4.2-arch1-1`` and I didnt have any issues. You should try 5.4 kernel.

enigma6174, 2019-12-17:
So is it possible to have Ubuntu 19.10 with `5.4.2-arch1-1` kernel? If yes, could you please send me the link for the same? The **ONLY** reason I am using Ubuntu is because of Deep Learning and the massive support it enjoys from all the developer communities, APIs and frameworks. And I chose 19.10 because Ubuntu 18 LTS has issues with the RTX series GPUs (For me, everything works butter smooth right out of the box albeit the lighting). I am worried changing my kernel might break support for deep learning or have issues with GPUs etc. and I can't sacrifice functionality for aesthetics.
All that said, it would be a welcome change to have everything working just like in Windows.

BogdanDarius, 2019-12-17:
You could update your kernel to 5.4.2, but you dont need the arch branded kernel.
I'm not sure it's because of the kernel, could be. Are the asus kernel modules loaded, what is the output of ```lsmod | grep asus```.
Here is mine:
```
asus_nb_wmi            28672  0
asus_wmi               36864  1 asus_nb_wmi
sparse_keymap          16384  1 asus_wmi
rfkill                 28672  10 asus_wmi,bluetooth,cfg80211
battery                24576  1 asus_wmi
wmi                    36864  3 asus_wmi,wmi_bmof,mxm_wmi
asus_wireless          20480  0
i8042                  32768  1 asus_nb_wmi
```

enigma6174, 2019-12-17:
> You could update your kernel to 5.4.2, but you dont need the arch branded kernel.
> I'm not sure it's because of the kernel, could be. Are the asus kernel modules loaded, what is the output of `lsmod [...]

I do not have access to my laptop right now so I will do it by tonight and post an update

CarwynNelson, 2019-12-18:
I'm not sure if I should raise this as a separate issue, so for now I'll chime in on here.

I think I have the opposite problem to many people here. I'm running ubuntu 19.10 and I want to _turn off_ the keyboard backlights but nothing seems to work. I've had this issue on 18.04 and 19.04.

Are there any pieces of debug information that I could post to help figure out why I can't turn the backlight off?

When I run `sudo ./rogauracore brightness 0` to turn the backlight off I get a `0` return code with no output and no change in the backlight. I have also tried to change the backlight to a static colour and that has the same result.

The output of `lsmod | grep asus` is:
```
asus_nb_wmi            28672  0
asus_wmi               32768  1 asus_nb_wmi
sparse_keymap          16384  1 asus_wmi
asus_wireless          20480  0
wmi                    32768  3 asus_wmi,wmi_bmof,mxm_wmi
video                  49152  2 asus_wmi,i915
```

I am running rogauracore-1.2.0 downloaded as a tar archive from source.

I am more than happy to do more complex debugging, but I might need some slight hand holding to get things going.

BogdanDarius, 2019-12-18:
@CarwynNelson What is your vendor ID and product ID?

CarwynNelson, 2019-12-19:
> @CarwynNelson What is your vendor ID and product ID?

This may sound like a daft question, but how do I find that out?

I can't believe I forgot to mention this in my original post but I am also running a Strix Scar III, would that result in a different vendor or product id?

BogdanDarius, 2019-12-19:
I actually own a Strix Hero III, but the laptops are almost identical.
You can use ```lsusb``` and search for ```ASUSTek Computer```.
Check if it is listed here
https://github.com/wroberts/rogauracore/blob/master/src/rogauracore.c#L452

CarwynNelson, 2019-12-19:
So I ran the command above and this is the output
```
➜  ~ lsusb | grep -i asus
Bus 001 Device 002: ID 0b05:1866 ASUSTek Computer, Inc. N-KEY Device
```
So I have to assume that my laptop is supported.

I'm going to see if I can compile with verbose mode on and see what output I get.

I've also noticed people posting about booting into windows and monitoring stuff on Wireshark. I'm more than happy to do that as I still have my windows partition about, but I would need a little guidance as Wireshark has always been a bit alien to me.

CarwynNelson, 2019-12-19:
So running `rogauracore` with verbose mode on nets the below output. I can't see anything that would indicate an issue.

```
➜  rogauracore git:(master) ✗ sudo ./rogauracore brightness 0
parse speed 0
args:
color1 48 7 172
color2 41 64 127
color3 0 0 192
color4 10 172 41
brightness 0
single_static
constructed 1 messages:
message 0: 5a ba c5 c4 00 00 00 00 00 00 00 00 00 00 00 00 00 
Initialising libusb
Initialised libusb.
Found 4 USB devices.
Checking device 1d6b:0003, address 1
Checking device 0b05:1866, address 2
Found ROG Aura Core keyboard.
Opened USB device.
Auto detach kernel mode set.
Got configuration descriptor.
Found 3 interfaces on the USB device.
Claimed interface 0.
Successfully sent all messages.
```
The below line wasn't giving me confidence that the keyboard device was being opened, so I made a small patch

```
# Before
Found ROG Aura Core keyboard.
```
```
# After
Found ROG Aura Core keyboard model 0b05:1866, address 2
```

So far everything looks correct to me.

JoshDreamland, 2019-12-20:
Hi, Strix Scar III is my machine (G731GW). Mind trying the changes in #26? They use a special command for rainbow. The brightness command should now be the same as in master, however.

CarwynNelson, 2019-12-30:
Sorry @JoshDreamland for the late reply.

I did write out a reply but for some reason it hasn't been submitted.

I tried out the changes in #26 and unfortunately it did not work, I get the same results. I've enabled verbose mode in your patch and I'm posting the output here to see if that proves useful at all.

I would be happy to do some packet sniffing under windows if that would help, but I would need a little bit of hand holding as that is quite new to me.

```
➜  rogauracore git:(retool) ✗ sudo ./rogauracore brightness 0
Reading argument 1 (brightness)
parse brightness 0
Parsed as 0
args:
[unspecified][unspecified][unspecified][unspecified][unspecified][unspecified][unspecified][unspecified]single_static
constructed 1 messages:
message 0: 5a ba c5 c4 00 00 00 00 00 00 00 00 00 00 00 00 00 
Initialising libusb
Initialised libusb.
Found 5 USB devices.
Checking device 1d6b:0003, address 1
Checking device 0b05:1866, address 3
Found ROG Aura Core keyboard.
Opened USB device.
Auto detach kernel mode set.
Got configuration descriptor.
Found 3 interfaces on the USB device.
Claimed interface 0.
Successfully sent all messages.
```

JoshDreamland, 2019-12-30:
Do any of the other commands work for you? `sudo rogauracore black` is another way of doing what you want, if the `single_static` message is understood by your keyboard.

CarwynNelson, 2019-12-30:
> Alternatively, if you have access to a Windows installation, you could do the same.

Could you point me towards that issue? Maybe I can leave some comments there.

I still have the default windows install around from ASUS so I'm more than happy to do some packet capture; I'm just not exactly sure how to do that. I'm assuming it involves wireshark but i've never quite got to grips with it.

Any basic steps to getting these packet captures would be greatly appreciated.

JoshDreamland, 2019-12-30:
The issue I was referring to is #23. The tool is also of no use on CJ's hardware, so I was thinking maybe it used a different protocol.

That said, while I wasn't paying attention, someone by the name of Luke showed up and posted a ton of Wireshark dumps in #33. Turns out that on Luke's machine, this tool works intermittently. I'm now thinking there is interference between asus-wmi and rogauracore, sometimes ending with only rogauracore functioning (as on Will's machine), and sometimes ending with only asus-wmi functioning (possibly what's happening on your machine, Luke's machine, or even CJ's machine).

I'll need more information before I can draw that conclusion. Do you have asus-wmi installed?

CarwynNelson, 2019-12-30:
> Do you have asus-wmi installed?

I wish it was that, but unfortunately I do not have asus-wmi installed to my knowledge. I wasn't actually aware it was a thing until your comment.

JoshDreamland, 2019-12-30:
It's a kernel module. `lsmod | grep asus` should list it if it's installed. Otherwise, `modprobe asus-wmi` would load it. If you actually don't have it installed, try installing it. It may support your keyboard natively.

CarwynNelson, 2019-12-30:
Looks like `asus_wmi` is enabled.

```
$ lsmod | grep asus
asus_nb_wmi            28672  0
asus_wmi               32768  1 asus_nb_wmi
sparse_keymap          16384  1 asus_wmi
asus_wireless          20480  0
wmi                    32768  3 asus_wmi,wmi_bmof,mxm_wmi
video                  49152  2 asus_wmi,i915
```
Before turning to rogauracore I was trying to use the built in LED controls to turn the backlight off, but that wasn't working so I have I am assuming that my device is not supported by asus-wmi?

JoshDreamland, 2019-12-30:
Safe to assume. It also doesn't support my device, but it doesn't interfere with rogauracore on my device, either. For me, it doesn't recognize any of the special keys on my keyboard except the volume keys, but it can control my display backlight, so between rogauracore and asus_wmi, I'm fully covered.

If you're comfortable doing so, I'd suggest blacklisting that module just long enough to see if rogauracore works without it. It's technically possible to screw something up doing this, so make sure you can boot into a plain TTY or have a rescue CD handy before you go blacklisting modules (you'll need to boot up and revert this change if something stops working). Anyway, to do this, just create a file in modprobe.d, eg, `/etc/modprobe.d/asus.conf`, and add `blacklist asus_wmi` to it. That should cause asus_wmi to not be loaded next reboot, which would tell us with certainty whether the module was interfering with rogauracore.

After you're done, just delete that new config file and things will go back to the way they are now. I'd recommend reverting it unless it turns out asus_wmi is doing more harm than good.

CarwynNelson, 2020-03-28:
I've been meaning to get back to this issue for a while and do some more debugging. I really have no idea why (i haven't updated rogauracore) but this has just randomly started working for me.

Maybe some underlying system update has made it work? I wish I could provide more information, and I will if I can - but this may well of resolved itself?

CarwynNelson, 2020-12-07:
It looks like the docs were updated with a new(?) command to "wake up" the keyboard.

```
sudo ./rogauracore initialize_keyboard
```

This worked perfectly for me and I am now able to use rogauracore to control my keyboard.