wroberts/rogauracore #36: Add SPEED paramter to Rainbow

saintlyzero, 2020-02-06:
Currently I can specify the value of `speed` parameter alongside `single_color` and `multi_color`<br> How to add the `speed` parameter alongside Rainbow and make it moving?

JoshDreamland, 2020-02-06:
The pull request I opened allows for optional arguments, but it uses a different protocol for the rainbow instruction. You could backport it to use the old rainbow instruction, but keep the optional speed argument.

saintlyzero, 2020-02-06:
Hey Josh,
Thanks for the quice response.

I tried backporting and running it, but when I execute `$ make`, it threw some errors.
> src/rogauracore.c:278:52: error: initializer element is not constant
     {"single_breathing", &single_breathing, 2, 1, {SPEED}},<br>
src/rogauracore.c:278:52: note: (near initialization for ‘FUNCTION_RECORDS[1].scalars[0]’)
src/rogauracore.c:279:54: error: initializer element is not constant
     {"single_colorcycle", &single_colorcycle, 0, 1, {SPEED}},<br>
src/rogauracore.c:279:54: note: (near initialization for ‘FUNCTION_RECORDS[2].scalars[0]’)
src/rogauracore.c:281:50: error: initializer element is not constant
     {"multi_breathing", &multi_breathing, 4, 1, {SPEED}},<br>
src/rogauracore.c:281:50: note: (near initialization for ‘FUNCTION_RECORDS[4].scalars[0]’)
src/rogauracore.c:291:44: error: initializer element is not constant
     {"brightness", &set_brightness, 0, 1, {BRIGHTNESS}},<br>
src/rogauracore.c:291:44: note: (near initialization for ‘FUNCTION_RECORDS[14].scalars[0]’)<br>
Makefile:395: recipe for target 'src/rogauracore.o' failed
make[1]: *** [src/rogauracore.o] Error 1

From [this](https://stackoverflow.com/questions/3025050/error-initializer-element-is-not-constant-when-trying-to-initialize-variable-w) thread, I figured out that declaring `const` over here is generating errors.
```C
const ScalarDef SPEED = { "SPEED", "speed", 1, 3 };
const ScalarDef BRIGHTNESS = { "BRIGHTNESS", "brightness", 0, 3 };
```
How do I fix this?

JoshDreamland, 2020-02-06:
Oh, I thought I patched that. I might still have local changes; I'll have to check later. The fix looks like what I did in #32.

saintlyzero, 2020-02-06:
Yes, replacing `const` with `#define` fixed it.<br>I've made a pull request, do check it. Now it compiles without any errors.<br>But still, I am not able to figure out how to add the `speed` parameter alongside rainbow.<br>I executed 
```bash
$ sudo rogauracore rainbow 2
``` 
and I'm getting `Function rainbow takes no arguments`.<br>Is this the right command?

JoshDreamland, 2020-02-06:
My master branch is behind wroberts', so the fix still applied to it, but the branch you wanted to edit was [retool](https://github.com/JoshDreamland/rogauracore/tree/retool/). But if the rainbow command Will is using works for you, the one I'm using likely will not. You'd need to plug the old rainbow code into it, as well, right around [line 298](https://github.com/JoshDreamland/rogauracore/blob/retool/src/rogauracore.c#L298).

saintlyzero, 2020-02-06:
Okay. I'll try the old rainbow code.

Thanks!