wroberts/rogauracore #38: Issues on GX701 / GX735

IThraen, 2020-02-17:
Hi, i'm not able to get rogauracore working on my archlinux install.
I'll be glad to help as i still have a Windows 10 + wireshark setup, all i need is some help figuring which packet are useful (i guess the usb out ones are those i must look for, but i don't know which parts of the packets are relevant).
I may help other issues as my keyboard is a 0b05:1866.

IThraen, 2020-02-17:
Ok, I've dig a bit, and it seems the initialization needs some tuning. I first thought i did a bad manipulation as rogauracore worked after rebooting from windows. But i can't change the colors after a cold boot. I'll be glad to help to solve this issue.

JoshDreamland, 2020-02-17:
Sometimes sending the wrong message to it causes it to stop responding to color change instructions. There may also be some kind of enable message, but I haven't found any evidence of one. I would suggest recording what happens when ArmouryCrate launches for the first time after a cold boot, and replying it on Linux (you can just use RogAuraCore to play back each message).

IThraen, 2020-02-17:
Oh ok i got it, i'll try this right now. But for the armoury part, do i need to stop the windows service and restart it? It seems that the color mode is set before login.

JoshDreamland, 2020-02-17:
That may work. If not, I'd try uninstalling it and reinstalling it. The dump format Wireshark uses is fine, or you can paste just the bytes to Pastebin or Gist.

IThraen, 2020-02-17:
Thanks a lot for your help. Here is the dump
[gx735_reload_armoury_crate.txt](https://github.com/wroberts/rogauracore/files/4216253/gx735_reload_armoury_crate.txt)
Looking at your other posts, do you have a Zephyrus variant too?

IThraen, 2020-04-11:
Hi, it looks like issue 33 does the job for me aswell! Nevertheless, the rainbow mode is still shutting down the keyboard.

closed 2020-04-11 by IThraen