wroberts/rogauracore #40: Ga502DU 0B05:1866

digizero, 2020-03-16:
I installed rogauracore after verifying support with the vendor id, but am unable to get it to do anything, no errors, just no lights.

the keyboard is a white light only keyboard, with the brightness controls on fn+up and down keys.

Let me know if I can get you any info from this device.

fatalgoth, 2020-03-20:
I've the same laptop with the same symptoms. I'm currently attempting to write a libusb userspace driver that works for this laptop, and will submit a pull request here once that's complete. Having never written one, it could take awhile though (starting from scratch so I hopefully don't miss anything).

Here's a packet dump from Windows where I mess with the LED buttons and then install Armory Crate at the end, in case anyone has any ideas.

[my_kb_windows.pcapng.zip](https://github.com/wroberts/rogauracore/files/4358076/my_kb_windows.pcapng.zip)

fatalgoth, 2020-03-21:
So I quickly realized that a driver for us would be better served in kernel space where most of the heavy lifting has already been completed. In fact, a small patch enables most of the function keys to work correctly, though on my laptop a few of the normal keys are being incorrectly mapped to the brightness controls:

```
diff --git a/drivers/hid/hid-asus.c b/drivers/hid/hid-asus.c
index e6e4c841fb06..6de0cf2d1de7 100644
--- a/drivers/hid/hid-asus.c
+++ b/drivers/hid/hid-asus.c
@@ -1040,6 +1040,8 @@ static const struct hid_device_id asus_devices[] = {
                USB_DEVICE_ID_ASUSTEK_ROG_KEYBOARD2), QUIRK_USE_KBD_BACKLIGHT },
        { HID_USB_DEVICE(USB_VENDOR_ID_ASUSTEK,
                USB_DEVICE_ID_ASUSTEK_ROG_KEYBOARD3), QUIRK_G752_KEYBOARD },
+       { HID_USB_DEVICE(USB_VENDOR_ID_ASUSTEK,
+               USB_DEVICE_ID_ASUSTEK_ROG_KEYBOARD4), QUIRK_USE_KBD_BACKLIGHT },
        { HID_USB_DEVICE(USB_VENDOR_ID_ASUSTEK,
                USB_DEVICE_ID_ASUSTEK_FX503VD_KEYBOARD),
          QUIRK_USE_KBD_BACKLIGHT },
diff --git a/drivers/hid/hid-ids.h b/drivers/hid/hid-ids.h
index 3a400ce603c4..e09fbc9f1e74 100644
--- a/drivers/hid/hid-ids.h
+++ b/drivers/hid/hid-ids.h
@@ -193,6 +193,7 @@
 #define USB_DEVICE_ID_ASUSTEK_ROG_KEYBOARD1 0x1854
 #define USB_DEVICE_ID_ASUSTEK_ROG_KEYBOARD2 0x1837
 #define USB_DEVICE_ID_ASUSTEK_ROG_KEYBOARD3 0x1822
+#define USB_DEVICE_ID_ASUSTEK_ROG_KEYBOARD4 0x1866
 #define USB_DEVICE_ID_ASUSTEK_FX503VD_KEYBOARD 0x1869
 
 #define USB_VENDOR_ID_ATEN             0x0557
diff --git a/drivers/hid/hid-quirks.c b/drivers/hid/hid-quirks.c
index 0e7b2d998395..08d0d2d4454d 100644
--- a/drivers/hid/hid-quirks.c
+++ b/drivers/hid/hid-quirks.c
@@ -297,6 +297,7 @@ static const struct hid_device_id hid_have_special_driver[] = {
        { HID_USB_DEVICE(USB_VENDOR_ID_ASUSTEK, USB_DEVICE_ID_ASUSTEK_ROG_KEYBOARD1) },
        { HID_USB_DEVICE(USB_VENDOR_ID_ASUSTEK, USB_DEVICE_ID_ASUSTEK_ROG_KEYBOARD2) },
        { HID_USB_DEVICE(USB_VENDOR_ID_ASUSTEK, USB_DEVICE_ID_ASUSTEK_ROG_KEYBOARD3) },
+       { HID_USB_DEVICE(USB_VENDOR_ID_ASUSTEK, USB_DEVICE_ID_ASUSTEK_ROG_KEYBOARD4) },
        { HID_USB_DEVICE(USB_VENDOR_ID_JESS, USB_DEVICE_ID_ASUS_MD_5112) },
        { HID_USB_DEVICE(USB_VENDOR_ID_TURBOX, USB_DEVICE_ID_ASUS_MD_5110) },
        { HID_BLUETOOTH_DEVICE(USB_VENDOR_ID_ASUSTEK, USB_DEVICE_ID_ASUSTEK_T100CHI_KEYBOARD) },

```
This patch doesn't yet enable the backlight LED controls either, but that looks like an initialization issue. Here's the message from dmesg:

```
[    4.507221] asus 0003:0B05:1866.0007: Asus failed to request functions: -75
[    4.507237] asus 0003:0B05:1866.0007: Failed to initialize backlight.
```

I'm going to go ahead and contact the Linux USB mailing list with a bunch more information on this, since it's now unrelated to this project.

fatalgoth, 2020-03-23:
Here's a patch for the latest stable kernel (5.5.11) that makes the backlight actually work, among other things. The only thing that doesn't work is the display brightness function keys, but I'm still working on that.

```
diff --git a/drivers/hid/hid-asus.c b/drivers/hid/hid-asus.c
index e6e4c841fb06..f7fa2ceb728b 100644
--- a/drivers/hid/hid-asus.c
+++ b/drivers/hid/hid-asus.c
@@ -260,12 +260,14 @@ static int asus_report_input(struct asus_drvdata *drvdat, u8 *data, int size)
 static int asus_event(struct hid_device *hdev, struct hid_field *field,
 		      struct hid_usage *usage, __s32 value)
 {
-	if ((usage->hid & HID_USAGE_PAGE) == 0xff310000 &&
+	/* No need to warn because it'll just spam the log until we get a proper fix.
+     *
+     * if ((usage->hid & HID_USAGE_PAGE) == 0xff310000 &&
 	    (usage->hid & HID_USAGE) != 0x00 &&
 	    (usage->hid & HID_USAGE) != 0xff && !usage->type) {
 		hid_warn(hdev, "Unmapped Asus vendor usagepage code 0x%02x\n",
 			 usage->hid & HID_USAGE);
-	}
+	}*/
 
 	return 0;
 }
@@ -415,13 +417,13 @@ static int asus_kbd_register_leds(struct hid_device *hdev)
 		return ret;
 
 	/* Get keyboard functions */
-	ret = asus_kbd_get_functions(hdev, &kbd_func);
+	/* ret = asus_kbd_get_functions(hdev, &kbd_func);
 	if (ret < 0)
-		return ret;
+		return ret;*/
 
 	/* Check for backlight support */
-	if (!(kbd_func & SUPPORT_KBD_BACKLIGHT))
-		return -ENODEV;
+	/*if (!(kbd_func & SUPPORT_KBD_BACKLIGHT))
+		return -ENODEV;*/
 
 	drvdata->kbd_backlight = devm_kzalloc(&hdev->dev,
 					      sizeof(struct asus_kbd_leds),
@@ -696,8 +698,8 @@ static int asus_input_mapping(struct hid_device *hdev,
 	if ((usage->hid & HID_USAGE_PAGE) == 0xff310000) {
 		set_bit(EV_REP, hi->input->evbit);
 		switch (usage->hid & HID_USAGE) {
-		case 0x10: asus_map_key_clear(KEY_BRIGHTNESSDOWN);	break;
-		case 0x20: asus_map_key_clear(KEY_BRIGHTNESSUP);		break;
+		//case 0x10: asus_map_key_clear(KEY_BRIGHTNESSDOWN);	break;
+		//case 0x20: asus_map_key_clear(KEY_BRIGHTNESSUP);		break;
 		case 0x35: asus_map_key_clear(KEY_DISPLAY_OFF);		break;
 		case 0x6c: asus_map_key_clear(KEY_SLEEP);		break;
 		case 0x7c: asus_map_key_clear(KEY_MICMUTE);		break;
@@ -755,8 +757,8 @@ static int asus_input_mapping(struct hid_device *hdev,
 		case 0xff0a: asus_map_key_clear(BTN_A);	break;
 		case 0xff0b: asus_map_key_clear(BTN_B);	break;
 		case 0x00f1: asus_map_key_clear(KEY_WLAN);	break;
-		case 0x00f2: asus_map_key_clear(KEY_BRIGHTNESSDOWN);	break;
-		case 0x00f3: asus_map_key_clear(KEY_BRIGHTNESSUP);	break;
+		//case 0x00f2: asus_map_key_clear(KEY_BRIGHTNESSDOWN);	break;
+		//case 0x00f3: asus_map_key_clear(KEY_BRIGHTNESSUP);	break;
 		case 0x00f4: asus_map_key_clear(KEY_DISPLAY_OFF);	break;
 		case 0x00f7: asus_map_key_clear(KEY_CAMERA);	break;
 		case 0x00f8: asus_map_key_clear(KEY_PROG1);	break;
@@ -1040,6 +1042,9 @@ static const struct hid_device_id asus_devices[] = {
 		USB_DEVICE_ID_ASUSTEK_ROG_KEYBOARD2), QUIRK_USE_KBD_BACKLIGHT },
 	{ HID_USB_DEVICE(USB_VENDOR_ID_ASUSTEK,
 		USB_DEVICE_ID_ASUSTEK_ROG_KEYBOARD3), QUIRK_G752_KEYBOARD },
+	{ HID_USB_DEVICE(USB_VENDOR_ID_ASUSTEK, 
+        USB_DEVICE_ID_ASUSTEK_ROG_KEYBOARD4), 
+        QUIRK_USE_KBD_BACKLIGHT | QUIRK_NO_CONSUMER_USAGES },
 	{ HID_USB_DEVICE(USB_VENDOR_ID_ASUSTEK,
 		USB_DEVICE_ID_ASUSTEK_FX503VD_KEYBOARD),
 	  QUIRK_USE_KBD_BACKLIGHT },
diff --git a/drivers/hid/hid-ids.h b/drivers/hid/hid-ids.h
index 3a400ce603c4..e09fbc9f1e74 100644
--- a/drivers/hid/hid-ids.h
+++ b/drivers/hid/hid-ids.h
@@ -193,6 +193,7 @@
 #define USB_DEVICE_ID_ASUSTEK_ROG_KEYBOARD1 0x1854
 #define USB_DEVICE_ID_ASUSTEK_ROG_KEYBOARD2 0x1837
 #define USB_DEVICE_ID_ASUSTEK_ROG_KEYBOARD3 0x1822
+#define USB_DEVICE_ID_ASUSTEK_ROG_KEYBOARD4 0x1866
 #define USB_DEVICE_ID_ASUSTEK_FX503VD_KEYBOARD	0x1869
 
 #define USB_VENDOR_ID_ATEN		0x0557
diff --git a/drivers/hid/hid-quirks.c b/drivers/hid/hid-quirks.c
index 0e7b2d998395..08d0d2d4454d 100644
--- a/drivers/hid/hid-quirks.c
+++ b/drivers/hid/hid-quirks.c
@@ -297,6 +297,7 @@ static const struct hid_device_id hid_have_special_driver[] = {
 	{ HID_USB_DEVICE(USB_VENDOR_ID_ASUSTEK, USB_DEVICE_ID_ASUSTEK_ROG_KEYBOARD1) },
 	{ HID_USB_DEVICE(USB_VENDOR_ID_ASUSTEK, USB_DEVICE_ID_ASUSTEK_ROG_KEYBOARD2) },
 	{ HID_USB_DEVICE(USB_VENDOR_ID_ASUSTEK, USB_DEVICE_ID_ASUSTEK_ROG_KEYBOARD3) },
+	{ HID_USB_DEVICE(USB_VENDOR_ID_ASUSTEK, USB_DEVICE_ID_ASUSTEK_ROG_KEYBOARD4) },
 	{ HID_USB_DEVICE(USB_VENDOR_ID_JESS, USB_DEVICE_ID_ASUS_MD_5112) },
 	{ HID_USB_DEVICE(USB_VENDOR_ID_TURBOX, USB_DEVICE_ID_ASUS_MD_5110) },
 	{ HID_BLUETOOTH_DEVICE(USB_VENDOR_ID_ASUSTEK, USB_DEVICE_ID_ASUSTEK_T100CHI_KEYBOARD) },

```

digizero, 2020-03-23:
Thanks @fatalgoth Worked like a charm, and my keyboard brightness function keys work as well. I patched the 5.6.0.1 RC instead of 5.5.

chrisj548, 2020-03-24:
Thank you @fatalgoth works great for me.

BenHall-1, 2020-04-28:
Okay because i'm completely new to kernel patching, how do you go about this? Sorry if I sound stupid...

Bobbydigital420, 2020-06-24:
@fatalgoth First of all thanks for this patch, works great! Just wondering if you've heard anything from the Linux mailing list about getting this patch merged? Would love to not have to compile it with every new kernel as I'm using a rolling distro.

fatalgoth, 2020-07-12:
@Bobbydigital420 Sorry for the late reply. This patch is pretty hacky but now that I'm less busy I'll try and make it kernel ready.