wroberts/rogauracore #42: how to switch off all the lights

amwink, 2020-03-31:
My laptop (_ASUS ROG Strix Scar II GL504GW-ES012T_ according to the order, keyboard _0xb05-0x1866_ according to `hardinfo`) is mainly for work use so I would like to turn the backlight and the ROG symbol on the lid off altogether.

After downloading and compiling `rogauracore` I ran
`$ rogauracore black`
and 
`$ rogauracore single_static 000000`
respectively (both followed by a reboot).

Unfortunately that leaves the backlight and the symbol in slowly varying 'rainbow' mode. According to the `readme.md` file this model would be supported so I had high expectations. Should I have done something differently or does `rogauracore` not work for this model (GW-ES012T)?

CarwynNelson, 2020-04-04:
I believe I have the same laptop model as you. I have been trying to get this working for quite a while now, and I recently discovered that rogauracore starts working if I boot into windows and then back into my ubuntu partition.

I'm not sure if this is of any help.

Also, if I do a full reboot back into linux it no longer works. Very strange.

MidhunSureshR, 2020-05-09:
Hello,

I see a lot of people having difficulty getting their backlight features working. If you are interested in reversing the backlight yourself using wireshark and viewing the actual URB packets that are being issued, you should review my [ powerpoint presentation](https://github.com/MidhunSureshR/openauranb/blob/master/slide.pptx) from my original project [repo](https://github.com/MidhunSureshR/openauranb).

Additionally the steps I used to find the device id is mentioned [here](https://github.com/MidhunSureshR/openauranb/issues/1#issuecomment-505950748).

closed 2023-02-03 by amwink