wroberts/rogauracore #43: Using on Arch

wantedrbk, 2020-04-09:
Hi can i use this for Manjaro KDE?

StaticRocket, 2020-04-14:
I used this on Arch just a few days ago and setup a package build earlier today.
https://aur.archlinux.org/packages/rogauracore-git/

MidhunSureshR, 2020-05-11:
It should work on all distributions (and even on Windows/Mac) as it is based on the cross-platform libusub library. This should be mentioned in the README.

MaikGit, 2020-06-13:
> I used this on Arch just a few days ago and setup a package build earlier today.
> https://aur.archlinux.org/packages/rogauracore-git/

Hi Static Rocket, i just installed your AUR package in my Manjaro linux OS and it works perfectly, the package is not on pacman and the programs control the keyboard perfectly.
I have a Scar 3 / G531GW