wroberts/rogauracore #44: can't control the led strip

shinigamiyadav, 2020-04-09:
i can't control the led strip below the keyboard can you add a separate option for the led strip i can't turn it on it will be highly appreciated. BTW @wroberts thanks for the project its amazing .

JoshDreamland, 2020-04-09:
The problem we're facing is that nearly-identical models of the same notebook and keyboard use entirely different messages to make the same requests. If you could get a Wireshark capture for changing the LED strip settings, it would be easier to add. Without that, no one can even begin to help. (The LED strip on my machine is controlled just fine.)

shinigamiyadav, 2020-04-14:
i turned off my led below my laptops keyboard and deleted windows now there is no way for me to capture the request. before deleting my windows i noticed that if i didn't turn off the led strip from windows it worked fine with this tool but then i made a mistake and forgot to turn it on before deleting windows. @JoshDreamland

JoshDreamland, 2020-04-14:
I see. In that case, it's possible one of us can add it based on our own system. I thought I had captured all the messages from my own machine, but I may have missed an option to individually toggle the lightstrip.

shinigamiyadav, 2020-04-15:
can you add it based on your machine i think it will work on mine too.