wroberts/rogauracore #45: Not working on GX501, different PID 0b05:1837

ekawahyu, 2020-04-12:
I found the PID from executing lsusb. When executing rogauracore, it says: Could not find ROG Aura Core keyboard.

JoshDreamland, 2020-04-12:
Does the tool work when you whitelist that value? A lot of these ASUS keyboards use completely distinct protocols for the same tasks.

ekawahyu, 2020-04-12:
how to put it into the whitelist?

JoshDreamland, 2020-04-12:
No, the ASUS_PRODUCT_IDS list, [here](https://github.com/wroberts/rogauracore/blob/master/src/rogauracore.c#L452). Your vendor ID is correct, but you need to add the product ID (0x1837) to that list.

ekawahyu, 2020-04-12:
Added to the list, no more error, but nothing happens. It maybe doing something but I don't seem to be able to control the brightness. Specific model of my machine is GX501VIK.

```
sudo rogauracore -v red
args:
color1 0 0 0
color2 0 0 0
color3 0 0 16
color4 247 104 250
single_static
constructed 1 messages:
message 0: 5d b3 00 00 ff 00 00 00 00 00 00 00 00 00 00 00 00 
Initialising libusb
Initialised libusb.
Found 6 USB devices.
Checking device 1d6b:0003, address 1
Checking device 8087:0a2b, address 5
Checking device 0b05:1837, address 4
Found ROG Aura Core keyboard.
Opened USB device.
Auto detach kernel mode set.
Got configuration descriptor.
Found 3 interfaces on the USB device.
Claimed interface 0.
Successfully sent all messages.
Sent SET message.
Sent APPLY message.
```

ekawahyu, 2020-04-12:
Too bad, my machine is not even listed here: https://www.asus.com/us/support/FAQ/1036506/

AleBasso80, 2020-04-22:
Hello,
I had the same problem, then I added the PID and the error disappeared, but the command didn't cause any change to the backlight that stayed off. However, if I first boot in Windows and then I restart (without shutdown) the PC in Ubuntu, then everything works fine. I had a look in the BIOS if I could enable the keyboard backlight at sturtup, but I didn't find anything.

Any suggestions?

Cheers
Alessandro

JoshDreamland, 2020-04-22:
@AleBasso80 We're discussing this issue in another thread. Check out [this comment](https://github.com/wroberts/rogauracore/issues/33#issuecomment-610060149).

AleBasso80, 2020-04-22:
@JoshDreamland sorry, I had missed that thread. I changed the source code as suggested and now everything is fine. Thank you for your help!