wroberts/rogauracore #46: Only works after a restart from windows on GA502DU

swagglepuf, 2020-05-02:
I only have brightness control on this model. It will only work after I do a restart from windows where the keyboard stays on. If I power the system off then back on and boot into manjaro, ubuntu, or fedora first. The commands never work and that includes running the sudo systemctl restart upower.service. 

This leads me to believe that the linux kernel cannot detect a keyboard back light on a cold boot. There for making this impossible to use on this model of pc unless you boot to windows do a restart and the load linux. 

There is a patch that has to be manually applied that brings full keyboard functionality. When using a rolling release you have to patch and compile a kernel ever 1-2 weeks and that gets old. Is there a way to implement what this patch does into this to allow for functionality with our requiring a boot into windows first. Here is the link the to actual patch that can be applied. 

https://patchwork.kernel.org/patch/9662291/

MidhunSureshR, 2020-05-09:
You might want to try and read [this](https://github.com/MidhunSureshR/openauranb/issues/1).
I think this project is based on the original code that I wrote for my ROG GL553VE model. This explains why it fails on a lot of other devices.

swagglepuf, 2020-05-09:
> You might want to try and read [this](https://github.com/MidhunSureshR/openauranb/issues/1).
> I think this project is based on the original code that I wrote for my ROG GL553VE model. This explains why it fails on a lot of other devices.

The interesting part is that if the keyboard light stays on when I boot into a system with this installed it works as intended. I can turn them off and set the 3 brightness levels using the rogauracore command. Is there a way to see what changes in my system when I do a cold boot vs a restart. Hopefully isolate the what the trigger is to get this functioning? Currently running manjaro, with one kernel patched so they function normally using the keyboard function key.

JoshDreamland, 2020-05-09:
This will probably fix it for you: https://github.com/wroberts/rogauracore/issues/33#issuecomment-610060149

swagglepuf, 2020-05-09:
> This will probably fix it for you: [#33 (comment)](https://github.com/wroberts/rogauracore/issues/33#issuecomment-610060149)

Going to be honest, I have no programming experience. None of that actually makes sense to me, I did try the forked version mentioned in the second to last and it doesnt work even though the post claims it to work.

closed 2020-05-10 by swagglepuf