wroberts/rogauracore #49: Rainbow turns off the RGB keyboard

WalterYeYint, 2020-05-15:
The command "sudo rogauracore rainbow" completely turns off the RGB keyboard on my laptop. 

Appreciate the effort of whoever are developing this.
Changed the RGB keyboard on my Asus ROG Strix Scar III to the new gold color and it looks so awesome.

cjkoral, 2020-05-21:
Try the changes in this PR for rainbow https://github.com/wroberts/rogauracore/pull/26

may need to initialize with the current rogaruacore first. 

The rainbow function in that fork worked for me

rajkush211, 2020-06-12:
Firstly, I'll appreciate the efforts of @wroberts for this, but in my case if i use rainbow then the RGB light is switching it to static red please fix this issue mate

WalterYeYint, 2021-05-07:
> Try the changes in this PR for rainbow #26
> 
> may need to initialize with the current rogaruacore first.
> 
> The rainbow function in that fork worked for me

Thanks
This works.