wroberts/rogauracore #58: not working ong GL503VS

kymbl, 2020-11-07:
[image]

My machine can't run it, I tried a couple of things but it didn't work.

wroberts, 2020-11-07:
@kymbl you'll need to do `make install` to get `rogauracore` on your path.  if you don't want to do that, you can just build the binary in the current directory with `make`, but then you need to give the relative path to the binary to run it: `./rogauaracore --help`.  Hope that helps!

kymbl, 2020-11-07:
[image]
If you mean this, I can't do it either.

kymbl, 2020-11-07:
I don't know how to use `make`, I would be so appreciated if you could teach me how to.

JoshDreamland, 2020-11-07:
Make is not a package manager; it only builds what is in the specified makefile (or, by default, the file named Makefile in the current directory).

Have you tried following the installation instructions from the readme displayed on the [main project page](https://github.com/wroberts/rogauracore)?

kymbl, 2020-11-08:
Okay, let me show you all the things I have done so far.
First of all, I have cloned the repository to my beloved pc, then downloaded the zip file and extracted it to the desktop.

[image]

Then, I actually just tried to copy-paste the `On Ubuntu from a release:` part, but there seem to be errors that I don't have a clue how to get rid of even though I have been struggling to understand why this error occurs for a day now.

[image]

wroberts, 2020-11-08:
The bit where it says `checking whether the C compiler works ... no` makes it look like either you don't have all the standard linux build tools installed or there is some kind of configuration problem with your setup.  You can try `sudo apt install --reinstall build-essential gcc` and retry `./configure` and `make`.  If that doesn't work you might try running `sudo dpkg-reconfigure build-essential` and `sudo dpkg-reconfigure gcc`.

kymbl, 2020-11-08:
Oh, can't say how much thankful I am. `sudo apt install --reinstall build-essential gcc` worked flawlessly. After reinstallation, the problem has solved. Thanks again.

wroberts, 2020-11-08:
Very cool, I'm glad you got things working :)

closed 2020-11-08 by kymbl