wroberts/rogauracore #69: Other Aura sync devices compatibility

xXLXx, 2021-01-24:
Is there a chance for this to work with other ROG Aura devices. I have an [ROG Throne](https://rog.asus.com/ph/headsets-audio/accessories/rog-throne-qi-model/) at home and just at least want to change the color to "not rainbow"

[image]

I managed to put in my device pid but it's not working, is there a way to know how different the messages then send an ROG Throne compared to the supported devices on this list?

Thanks

JoshDreamland, 2021-01-24:
Do you have a Windows installation available? All of the functionality we've added so far has been done by capturing messages sent by the official software using WireShark. As far as I know, no one's reverse-engineered ArmouryCrate to extract the messages from it directly,  but especially if the messages are indexed in the PE header, that might be a viable option.

xXLXx, 2021-01-24:
Ahh I see. And yes I use the one from armoury crate when I installed this on my windows machine: https://dlcdnets.asus.com/pub/ASUS/mb/14Utilities/ArmouryCrateInstallTool.zip