wroberts/rogauracore #72: A gui for rogauracore

rastafaninplakeibol, 2021-04-03:
Hi man, first of all thank you a lot for your fantastic tool, it is really really really helpful and finally i'm able to use my asus keyboard on linux too. Some time ago i made a gui in electron to use your tool easily. I don't know if you are interested or not, but take a look at my repo https://github.com/rastafaninplakeibol/rogauracore-gui

If you think it can be helpful feel free to cite my repo, somewhere, or if you have some advices i'll be glad to hear them :)

wroberts, 2021-08-16:
@rastafaninplakeibol looks quite promising.  can you make a screenshot of your GUI and embed it into the README of your repo?

rastafaninplakeibol, 2021-09-06:
Hello, sorry for the delay, I've just updated my README. There isn't a great description on how to use it because it seems quite intuitive to me, but if you encounter any problem just tell me ^^