wroberts/rogauracore #78: Could not find ROG Aura Core keyboard.

donkikhot, 2021-07-27:
Pleas add support ROG Zephyrus M16

JoshDreamland, 2021-07-27:
This isn't something we can do alone without purchasing a Zephyrus M16 and configuring it. If you are willing to try plugging in your device ID [to this array](https://github.com/wroberts/rogauracore/blob/master/src/rogauracore.c#L469) (you can obtain it using lsusb), you should be able to test out the application for your keyboard. If it works, please send a pull request to add it permanently so others with your model can use it, too.

donkikhot, 2021-07-27:
The given id already exists in your array.
Bus 003 Device 004: ID 0b05: 19b6 ASUSTek Computer, Inc. N-KEY Device

JoshDreamland, 2021-07-27:
Should be supported, then. Are you running the `rogauracore` tool as root?

donkikhot, 2021-07-27:
Yes...
 sudo rogauracore initialize_keyboard
[sudo] пароль для don: 
Could not find ROG Aura Core keyboard.

MIBbrandon, 2021-07-29:
I have the same problem on the Zephyrus G15:
Bus 001 Device 002: ID 0b05:19b6 ASUSTek Computer, Inc. N-KEY Device

wroberts, 2021-07-29:
@MIBbrandon @donkikhot can you run `sudo rogauracore -v red` please?

MIBbrandon, 2021-07-29:
args:
color1 0 128 0
color2 0 0 0
color3 0 0 248
color4 173 160 139
single_static
constructed 1 messages:
message 0: 5d b3 00 00 ff 00 00 00 00 00 00 00 00 00 00 00 00 
Initialising libusb
Initialised libusb.
Found 7 USB devices.
Checking device 1d6b:0003, address 1
Checking device 27c6:521d, address 2
Checking device 1d6b:0002, address 1
Checking device 1d6b:0003, address 1
Checking device 13d3:3563, address 3
Checking device 0b05:19b6, address 2
Checking device 1d6b:0002, address 1
Could not find ROG Aura Core keyboard.

donkikhot, 2021-07-30:
sudo rogauracore -v red
args:
color1 72 176 16
color2 41 252 127
color3 0 0 248
color4 253 83 97
single_static
constructed 1 messages:
message 0: 5d b3 00 00 ff 00 00 00 00 00 00 00 00 00 00 00 00 
Initialising libusb
Initialised libusb.
Found 8 USB devices.
Checking device 1d6b:0003, address 1
Checking device 27c6:521d, address 3
Checking device 13d3:5458, address 2
Checking device 13d3:3563, address 5
Checking device 0b05:19b6, address 4
Checking device 1d6b:0002, address 1
Checking device 1d6b:0003, address 1
Checking device 1d6b:0002, address 1
Could not find ROG Aura Core keyboard

donkikhot, 2021-07-31:
Hi
Works well on Fedora 3.4.

LSB Version:	:core-4.1-amd64:core-4.1-noarch
Distributor ID:	Fedora
Description:	Fedora release 34 (Thirty Four)
Release:	34
Codename:	ThirtyFour

kernel: 5.13.5-200.fc34.x86_64

JoshDreamland, 2021-08-02:
Were you running an outdated version on your previous OS, somehow...?

donkikhot, 2021-08-02:
No, before that I used ubuntu with a 5.13 kernel, installed as

`sudo apt install libusb-1.0-0 libusb-1.0-0-dev
VERSION=1.4
curl -LOs https://github.com/wroberts/rogauracore/releases/download/$VERSION/rogauracore-$VERSION.tar.gz
tar xf rogauracore-$VERSION.tar.gz
cd rogauracore-$VERSION/
./configure
make
sudo make install`

Installed on Fedora like this
`autoreconf -i
./configure
make`

At the moment everything is working, but there are problems with the brightness adjustment from the hotheys. Also, during sleep, the keyboard remains lit.

wroberts, 2021-08-16:
@MIBbrandon any updates on your end?

tsukikage, 2024-08-15:
Having this same problem with my ASUS TUF Gaming A15 FA507NU on Ubuntu 24.04 LTS.

(base) tsukikage@tachikoma:~$ sudo apt install libusb-1.0-0 libusb-1.0-0-dev
[sudo] password for tsukikage: 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
libusb-1.0-0 is already the newest version (2:1.0.27-1).
libusb-1.0-0 set to manually installed.
The following NEW packages will be installed:
  libusb-1.0-0-dev libusb-1.0-doc
0 upgraded, 2 newly installed, 0 to remove and 4 not upgraded.
Need to get 277 kB of archives.
After this operation, 2,134 kB of additional disk space will be used.
Get:1 http://us.archive.ubuntu.com/ubuntu noble/main amd64 libusb-1.0-0-dev amd64 2:1.0.27-1 [77.7 kB]
Get:2 http://us.archive.ubuntu.com/ubuntu noble/main amd64 libusb-1.0-doc all 2:1.0.27-1 [199 kB]
Fetched 277 kB in 2s (169 kB/s)          
Selecting previously unselected package libusb-1.0-0-dev:amd64.
(Reading database ... 175554 files and directories currently installed.)
Preparing to unpack .../libusb-1.0-0-dev_2%3a1.0.27-1_amd64.deb ...
Unpacking libusb-1.0-0-dev:amd64 (2:1.0.27-1) ...
Selecting previously unselected package libusb-1.0-doc.
Preparing to unpack .../libusb-1.0-doc_2%3a1.0.27-1_all.deb ...
Unpacking libusb-1.0-doc (2:1.0.27-1) ...
Setting up libusb-1.0-doc (2:1.0.27-1) ...
Setting up libusb-1.0-0-dev:amd64 (2:1.0.27-1) ...

(base) tsukikage@tachikoma:~$ git clone https://github.com/wroberts/rogauracore.git
Cloning into 'rogauracore'...
remote: Enumerating objects: 271, done.
remote: Counting objects: 100% (61/61), done.
remote: Compressing objects: 100% (27/27), done.
remote: Total 271 (delta 46), reused 38 (delta 33), pack-reused 210 (from 1)
Receiving objects: 100% (271/271), 57.41 KiB | 286.00 KiB/s, done.
Resolving deltas: 100% (133/133), done.

(base) tsukikage@tachikoma:~$ cd rogauracore

(base) tsukikage@tachikoma:~/rogauracore$ sudo apt install build-essential
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
build-essential is already the newest version (12.10ubuntu1).
build-essential set to manually installed.
0 upgraded, 0 newly installed, 0 to remove and 4 not upgraded.

(base) tsukikage@tachikoma:~/rogauracore$ sudo apt install autoconf
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following additional packages will be installed:
  automake autotools-dev m4
Suggested packages:
  autoconf-archive gnu-standards autoconf-doc libtool gettext m4-doc
The following NEW packages will be installed:
  autoconf automake autotools-dev m4
0 upgraded, 4 newly installed, 0 to remove and 4 not upgraded.
Need to get 1,186 kB of archives.
After this operation, 4,463 kB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 http://us.archive.ubuntu.com/ubuntu noble/main amd64 m4 amd64 1.4.19-4build1 [244 kB]
Get:2 http://us.archive.ubuntu.com/ubuntu noble/main amd64 autoconf all 2.71-3 [339 kB]
Get:3 http://us.archive.ubuntu.com/ubuntu noble/main amd64 autotools-dev all 20220109.1 [44.9 kB]
Get:4 http://us.archive.ubuntu.com/ubuntu noble/main amd64 automake all 1:1.16.5-1.3ubuntu1 [558 kB]
Fetched 1,186 kB in 1s (1,276 kB/s)
Selecting previously unselected package m4.
(Reading database ... 175655 files and directories currently installed.)
Preparing to unpack .../m4_1.4.19-4build1_amd64.deb ...
Unpacking m4 (1.4.19-4build1) ...
Selecting previously unselected package autoconf.
[... 152 lines cut ...]
   rainbow_cycle
   red
   green
   blue
   yellow
   gold
   cyan
   magenta
   white
   black
   rainbow
   brightness
   initialize_keyboard

Function single_static takes 1 color(s):
   rogauracore single_static COLOR1 

COLOR argument(s) should be given as hex values like ff0000
SPEED argument should be given as an integer: 1, 2, or 3

(base) tsukikage@tachikoma:~$ rogauracore initialize_keyboard
Could not find ROG Aura Core keyboard.

(base) tsukikage@tachikoma:~$ sudo rogauracore initialize_keyboard
Could not find ROG Aura Core keyboard.