wroberts/rogauracore #81: Could not find ROG Aura core keyboard - Asus TUF FA56II

PrathamBhatTech, 2021-08-15:
I followed the installation instructions exactly.
after completing the installation any commands i run like
`sudo rogauracore single_static 0000ff`
`sudo rogauracore single_static 00ff00`
`sudo rogauracore single_static ffff00`
`sudo rogauracore multi_static ff0000 ffff00 00ff00 00ffff`
`sudo rogauracore single_colorcycle 1`
all return `Could not find ROG Aura Core keyboard.`

I tried `sudo rogauracore initialize_keyboard` and `sudo systemctl restart upower.service` but no luck.
I really want this to work cause i dont want to boot to windows just to do something as simple as changing color or cycle mode.

Ubuntu 20.04
Asus TUF FA56II

PrathamBhatTech, 2021-08-15:
I can change the brightness of the keyboard using the inbuilt keyboard shortcuts but not the cycle of breathing, static etc

wroberts, 2021-08-16:
hey @PrathamBhatTech thanks for the ticket.  the next step here is to find the device ID of your keyboard:
```
$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 0b05:1869 ASUSTek Computer, Inc. ITE Device(8910)
Bus 001 Device 004: ID 13d3:5666 IMC Networks USB2.0 HD UVC WebCam
Bus 001 Device 003: ID 8087:0a2b Intel Corp. Bluetooth wireless interface
Bus 001 Device 002: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
so for me that's `0b05:1869`.

You can try putting that ID in the [list of supported devices](https://github.com/wroberts/rogauracore/blob/master/src/rogauracore.c#L469), recompiling, and see if anything changes for you.

tbeedril, 2021-08-16:
I have the same problem and i't no keyboard in this list but my laptop has it
```
$ sudo lsusb
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 002: ID 13d3:56a2 IMC Networks USB2.0 HD UVC WebCam
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 13d3:3530 IMC Networks Bluetooth Radio 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

jexxadox, 2021-12-04:
any solution? my Asus Tuf Gaming A17 laptop won't initialize. the backlit keyboard was one of the major selling points!

PrathamBhatTech, 2022-01-06:
> hey @PrathamBhatTech thanks for the ticket. the next step here is to find the device ID of your keyboard:
>
> ```
> $ lsusb
> Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
> Bus 001 Device [...]

Hey @wroberts
```
$ lsusb
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 002: ID 04ca:4005 Lite-On Technology Corp. Bluetooth Radio
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 13d3:56a2 IMC Networks USB2.0 HD UVC WebCam
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
I think the keyboard is not listed? but I'm able to use it.

PrathamBhatTech, 2022-01-06:
Sorry for the late reply. I missed the notification for your reply. And forgot about this.

wroberts, 2022-01-06:
No worries.  So, yeah, that listing doesn't seem to contain a keyboard entry; the keyboard must be connected on some other interface.  

Unfortunately, for @jexxadox @PrathamBhatTech @tbeedril I think this is the end of the road for your experiment with `rogauracore`, because that's just a client that uses the USB bus to try to duplicate the protocol from Asus's aura core app.  No USB, no worky.  You might have better luck with [OpenRGB](https://openrgb.org/).

javierisassi, 2022-09-21:
I added my id: Bus 001 Device 014: ID 0b05:193c ASUSTek Computer, Inc. ROG FALCHION to the mix, rebuild now
using sudo rogauracore takes my command (no errors) but the keyboard doesn't change colors or respond to the commands. Oh well.

muerfox, 2024-09-10:
hello!
i do have a same issue in rog flow z13
the lsusb shows when keyboard reconnects as below
Bus 003 Device 011: ID 0b05:1a30 ASUSTek Computer, Inc. N-KEY Device
the default backlight starts when i reconnect the keyboard after boot but still cant use rogauracore the output only shows
Could not find ROG Aura Core keyboard.