wroberts/rogauracore #82: Keyboard has suddenly gone dark on GX701GX-XB78

lehthanis, 2021-09-12:
I know my keyboard lighting is good because it turns red when I boot into windows.  But after a recent archlinux update, my keyboard has gone dark.  When I run rogauracore red the keypad lighting on the touchpad will flash off and then on again if it's on, but that's all. What could have happened and is there anything I can do to fix it?

running with -v gives me this:
```
args:
color1 0 1 0
color2 0 0 0
color3 0 0 64
color4 0 0 0
single_static
constructed 1 messages:
message 0: 5d b3 00 00 ff 00 00 00 00 00 00 00 00 00 00 00 00 
Initialising libusb
Initialised libusb.
Found 7 USB devices.
Checking device 1d6b:0003, address 1
Checking device 0b05:1866, address 3
Found ROG Aura Core keyboard.
Opened USB device.
Auto detach kernel mode set.
Got configuration descriptor.
Found 3 interfaces on the USB device.
Claimed interface 0.
Sent INIT messages.
Successfully sent all messages.
Sent SET message.
Sent APPLY message.
```

lsusb gives me:
```
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 0b05:1866 ASUSTek Computer, Inc. N-KEY Device
Bus 001 Device 002: ID 1532:0062 Razer USA, Ltd Atheris
Bus 001 Device 004: ID 8087:0aaa Intel Corp. Bluetooth 9460/9560 Jefferson Peak (JfP)
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

sptls, 2021-09-12:
Same thing happened to me. Stopped working after last system update

lehthanis, 2021-09-12:
What's interesting is that before I had the software my keyboard just stayed lit up in whatever configuration was last set or factory default. Now it actually shuts off when I boot into Linux and turns on when I boot into windows. Odd

sptls, 2021-09-16:
> What's interesting is that before I had the software my keyboard just stayed lit up in whatever configuration was last set or factory default. Now it actually shuts off when I boot into Linux and turns on when I boot into windows. Odd

Actually tried change brightness today and it worked... 
Try `rogauracore brightness 3`

lehthanis, 2021-09-16:
> I

HOLY CRAP!  It worked!!!  Thanks much!

sptls, 2021-09-16:
That was very silly and weird, because I remember I didn't mess with brightness settings. 
Anyway; glad I could help ;)

sptls, 2021-09-19:
> > I
> 
> HOLY CRAP! It worked!!! Thanks much!

I was bored today and made quick GUI wrapper for this program, you can check it here:
https://github.com/sptls/rogauracore_gui

PR #83 (2021-11-11): Update README.md according to #82
If the keyboard's lights default to a brightness of 0 when booting into Linux, the user might mistakenly think their commands are not correctly sent or interpreted. This confusion is especially likely to occur if the brightness defaults to a higher value when booting into e.g. Windows. I've updated the README to instruct users to try the brightness command in case their keyboard remains dark. 

Closes #82
commit 93d714a1 (2021-11-11): Update README.md according to #82

If the keyboard's lights default to a brightness of 0 when booting into Linux, the user might mistakenly think their commands are not correctly sent or interpreted. This confusion is especially likely to occur if the brightness defaults to a higher value when booting into e.g. Windows. I've updated the README to instruct users to try the brightness command in case their keyboard remains dark. 

Closes #82
closed 2021-11-11 by wroberts