wroberts/rogauracore #90: Not working on Zephyrus duo 16 (2023)

EKami, 2023-05-11:
When I try to use it on my GX650PY I get:
```
$ rogauracore single_static ff0000
Could not open ROG Aura Core keyboard: LIBUSB_ERROR_ACCESS
```

Any idea how to fix this?

mxkov, 2023-05-14:
Did you set the brightness level? E.g.
`$ sudo rogauracore brightness 2`

EKami, 2023-05-14:
It's working lol, thank you very much!

closed 2023-05-14 by EKami

EKami, 2023-05-14:
@mxkov 
Any idea why it doesn't work if I run it in a bash script?

```
#!/bin/bash

set -e
rogauracore rainbow_cycle 1
rogauracore brightness 2
```

I want to run those commands when my session starts. Thanks!

mxkov, 2023-05-18:
@EKami 
Personally, I set up a service to automatically run rogauracore at startup. Here's how you can do it.

1) Save your script to /usr/local/sbin/, but start every ```rogauracore``` command with ```sudo```. Let's say you saved your script to  /usr/local/sbin/rogauracore-auto.sh 
2) Edit permissions:
```
sudo chmod 0700 /usr/local/sbin/rogauracore-auto.sh
 ```
3) Make file /etc/systemd/system/rogauracore-auto.service with the following content:
```
[Unit]
Description=Enable ASUS RGB backlight (or whatever description you want)
[Service]
ExecStart=/usr/local/sbin/rogauracore-auto.sh
[Install]
WantedBy=multi-user.target
```
4) Then run:
```
systemctl start rogauracore-auto.service
systemctl enable rogauracore-auto.service
```

Hope this works for you

EKami, 2023-05-19:
This is also exactly what I did but my script had this:
```
#!/bin/bash

set -e

rogauracore rainbow_cycle 1
sleep 2
rogauracore brightness 2
```

As soon as I removed `rogauracore rainbow_cycle 1` from the script it started working. For some reason, the command does not work with two consecutive calls even with a `sleep` in the middle.